vlachoudis/bCNC #784: Usb connector

avilab84, 2018-03-23:
the USB button shows a blue cable that was used in the 90s. 
I would prefer a icon of a usb cable.
https://github.com/vlachoudis/bCNC/blob/master/icons/serial32.gif
[image]

vlachoudis, 2018-03-26:
Correct. Could you find a nice one with 32x32 pixels. When I resize it the result is not nice.

masterleo, 2018-03-26:
Maybe this one ? 

[image]

masterleo, 2018-03-26:
I use it on my BCNC it is great 
[image]

masterleo, 2018-03-27:
On the same topic, the button should be green when connected and red when disconnected. (Quote from my girlfriend) The actual behaviour shows red when connected.

vlachoudis, 2018-04-16:
done

closed 2018-04-16 by vlachoudis