vlachoudis/bCNC #786: Interrupting

Quiigi, 2018-03-25:
Is there a way to abort a long-running operation?

I'm learning bCNC from the wiki and by trying things out.  Specifically, I was experimenting with Profile on a simple shape; I chose the outline of letter "g" in Times_New_Roman.  Choosing "outside" works fine, it completes in a split second and the result looks correct.  But choosing "inside" runs forever.  The "phython" process consumes 100% CPU and ever more memory (currently about 12G), till hours later the OOM killer terminates it.  So clearly I don't understand the "inside" profile.

In a situation like that, I'd like to end this experiment.  Terminating bCNC (e.g., with C-c or "kill") takes care of it, but I also lose anything that's not saved from the current session (gcode, materials, end mills).
Is there a gentler way to tell bCNC to give up on the thing I asked for, something with the same effect as this idiom that plugins use when they decide to give up:

`app.setStatus(_("Foo abort: This plugin requires Bar"))
return`

vlachoudis, 2018-03-26:
Can you send me the example that went to infinite loop.

Quiigi, 2018-04-06:
I'm not too worried that my "profile" failed -- I asked bCNC to do something that didn't make sense.  I was more interested in a way to stop a run-away operation without killing bCNC altogether.
   ~~~
Anyway, here's what I did:
Start bCNC.
Click Tools > CAM > Generator > Text
Enter:
-Text to generate "g"
-Font file "/usr/share/fonts/truetype/msttcorefonts/Times_New_Roman.ttf"
-Font size 10.0

Leave other inputs default (Working Depth 0.0, Close Contours enabled, Image to ASCII empty).
Click on title "Text" to generate outline path of letter "g".
[text-g.gcode.txt](https://github.com/vlachoudis/bCNC/files/1885669/text-g.gcode.txt

Click on the path to select it.
Still in Tools, click "Profile".
Choose End Mill "Bit1" (it comes with bCNC, radius = 3.175mm = 1/8 inch),
Direction "inside".  Leave Additional offset distance = 0.0.

If I leave the Font size at 30.0, the profile operation completes instantly, with a correct result.
I suspect the problem is that my result (on the smaller letter) is empty.  As far as I can tell, the loop happens where "profile" invokes "opath.intersectSelf()", at https://github.com/vlachoudis/bCNC/blob/master/CNC.py#L3526 (presumably the line number could change if there are unrelated changes above it in the same file).