vlachoudis/bCNC #789: Please add simple math features to setting WPos (or any other numeric input box)

Harvie, 2018-03-28:
Hello, it would be super useful if it was possible to manual set WPos not only as number, but also as expression. Especialy dividing by two is very useful to find center of holes, etc... So i can set WPos to `10/2` instead of just `5`. I know that i can open external calc program and compute it myself, but this would be much more user friendly. This might be useful in most of numeric inputs in bCNC, but especially in WPos box.

How it would work:

  1. find one edge of the part
  2. zero WPos
  3. jog to oposed edge of part
  4. append `/2` to WPos and hit enter
  5. move to the origin (BAM! you are in the center of the part)
  6. possibly do the same for other axes

You can see the example in this video: https://youtu.be/nkYpKWbA2P0?t=300

vlachoudis, 2018-03-28:
@Harvie  this is already possible in the command line
you can set any function there using the set command
e.g.
```set 0 [10/2] [cos(pi/3)]```
or only to individual components like
```setx [(xmin+xmax)/2]```
Something that I use quite often is
```set [xmin] [ymin]```
to set to the min x,y of the active gcode
there are several internal variables, like the margins of the plot,...

vlachoudis, 2018-03-28:
I see the point. It is not a problem in implementing this

Harvie, 2018-07-02:
There seems to be some similar functionality in `ControlPage.py`. I am not near the machine to test if it fits my needs. It doesn't do anything when not connected to machine. I've found this:

```python
        def setX(self, event=None):
                if self.app.running: return
                try:
                        value = float(eval(self.xwork.get(),CNC.vars,self.app.gcode.vars))
                        self._wcsSet(value,None,None)
                except:
                        pass
```

Harvie, 2019-01-09:
I need to document this feature and then i will close the issue.

atrueresistance, 2019-01-09:
I had no idea this existed in bCNC!

Harvie, 2019-01-16:
I've added even better math support to DRO:

[image]

Harvie, 2019-01-16:
I've mentioned this feature in wiki, so i think this can be closed.
https://github.com/vlachoudis/bCNC/wiki/Jogging

closed 2019-01-16 by Harvie