vlachoudis/bCNC #792: Moving origin garbles g-code

gyakubov, 2018-03-30:
Hi there.
First off - very nice project - really great job.
Not without issue, though...

When I am trying to move origin of the loaded g-code in the editor it garbles in a way:
was
G0X10Y10Z10
became
G0X20Y20Z10 G0X20Y20Z10

In another words - each code after  transformation repeated twice in line.

Quiigi, 2018-04-07:
I agree on both:  bCNC is awesome.  And I confirm that moving doubles some of the g-code.

I'd like to add a detail: when I move _objects_ (key "m" or blue arrows "Move"), it doubles the coordinates.  But in contrast to Gyakubov's findings, it does not duplicate the the instruction, e.g., G0.
In my case,
~g1 x4.372656 y5.562109~
becomes
~G1 X0.346836 Y2.283349 X0.346836 Y2.283349~
(Text generated "g1", Move rewrote that as "G1")

The doubling of coordinates happens in the block starting at https://github.com/vlachoudis/bCNC/blob/master/CNC.py#L3731 
~
					for c in check:
						try:
							if c not in present and new.get(c) != old.get(c):
								newcmd.append(self.fmt(c,new[c]/self.cnc.unit))
						except:
							pass
~
In my case, check == "XYZ".  The variable present is initialized to empty at https://github.com/vlachoudis/bCNC/blob/master/CNC.py#L3717 and never changed, hence the first condition is trivially true, and if the coordinate changed at all, L3734 repeats what L3721 (identical code) already did.

vlachoudis, 2018-04-07:
Can you please download the latest master, ive fixed that some days ago

Quiigi, 2018-04-07:
Thanks, yes, I did pull your master.
After looking into the code, I noticed you had fixed it, hence I put at the top "This may be a duplicate", and at the bottom that your commit fixed it.  I'm changing my comment to make this clearer.

closed 2018-04-07 by vlachoudis