vlachoudis/bCNC #794: Opening error more than 10000 lines of G-code

sfinexer, 2018-03-31:
When opening G-code of more than 10000 lines, display (visual) on the screen doesn't happen!
Also there is no korrektry continuation after a pause if more than 10000 lines of G-code are passed.

atrueresistance, 2018-03-31:
I've run programs well over 750,000 lines. When there is something that large the display won't update, but will send the code perfectly fine. On the right side there is a dropdown with some numbers in it, that will tell bCNC how long to attempt to render before timing out. 

I can't really say about the pausing, as I just let it run mostly or break out my operations into segments that I can run when I'm near the machine.

sfinexer, 2018-03-31:
Whether it is possible to make display?
The G-code teams work, however the start after a pause on a large number of the lines G-code sometimes goes not truly.

Harvie, 2018-07-19:
Related to #591

closed 2018-09-10 by sfinexer