vlachoudis/bCNC #797: V0.9.10 Slow feeds pauses grbl

chepo92, 2018-04-10:
Have some godes hanging/pausing grbl with an error when setting a low speed on gcode or trying to override for slows speed.
Example (gcode cuts out a rectangle pcb of 2mm thikness, at 10mm/min)
M03 S8000
G00 F3000
G00 Z2
G00 X-1.79 Y1.79
G00 Z0
G01 F60 Z-2
executes ok, then the following
G01 F10 X-81.65  pauses some mm after starting to move to X-81.65
G01 Y63.49
G01 X-1.79
G01 Y1.79
G00 Z2
M05
M02

With feeds over 15mm/min, and no overrride the issue doesn't occur (executes ok)
Sending gcodes manually (through terminal) also execute ok.
with feed overrides slower than 10mm/min the isssue occurs. 

I think it's an issue with the communication/buffering bcnc-grbl, as grbl sends an ok when receives gcode command, but it doesn't have enough time to execute movement, then grbl pauses as bcnc is sending next command before the machine has done previous move

[pcb.txt](https://github.com/vlachoudis/bCNC/files/1892958/pcb.txt)

vlachoudis, 2018-04-16:
With grbl 0.9 or 1.1?

chepo92, 2018-04-16:
It's grbl 1.1

chepo92, 2018-04-19:
Ok read grbl wiki again and its a limitation of arduino 
https://github.com/gnea/grbl/wiki/Known-Issues#doesnt-go-slower-than-x-mmmin-or-limitations-at-very-slow-step-rates

My minimum possible fr is 2.25, so when i set a slow feed under 2mm/s (gcode or feed override) arduino just hangs and restart

closed 2018-04-19 by chepo92