vlachoudis/bCNC #802: z-axis limit nightmare

def670, 2018-04-12:
Hi. I am having some issues with GRBL/bCNC on my small machine.  It is set up as a plotter with a pen, and mostly seems to work right, however the z-axis limit is giving me lots of grief.  
I can get the system to home fine.  when I have it draw the pen always goes up past the upper limit on z.  The max travel for z is set at 3mm, but it travels past 12mm to hit the limit EVERY FREAKING TIME!  Starting to give me mental jitters.

$I
[VER:1.1f.20170801:]
[OPT:V,15,128]
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok

$$
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=500.000
$26=25
$27=0.750
$30=1000
$31=0
$32=0
$100=123.076
$101=145.454
$102=533.333
$110=500.000
$111=500.000
$112=200.000
$120=10.000
$121=10.000
$122=10.000
$130=145.000
$131=176.000
$132=3.000
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok

the limit for z is attached to pin 12 on my arduino
I am using a genuine arduino uno
this is not an issue of electrical noise - the pen is lifting approx 4x higher than it should be able to due to $132.  definatly actually triggering the limit

def670, 2018-04-12:
in my current drawing I am doing a gear from the tools submenu.  The gear draws, then the pen rises and triggers the limit.   The gear is not drawing correctly but i believe that is due to an issue with a bad belt and is not part of this problem I am working on.
thanks in advance

wwjohb, 2018-04-12:
Does the machine have a limit switch that gets triggered on a normal lift of the pen? That doesn't sound quite right. It should only get triggered if it goes too high, ie. ***if*** z axis is 30 then a pen lift should be able to start at -25 below the limit come up to -22 (up 3 mm because $13=0) then go back down without triggering the limit switch. If its up near -1 (within a few mm of the limit switch at the top of the Z axis) and triggers the switch that would be the correct operation. How is the machine setup? Picture and the G-Code used may help, default gear generator GCode only lifts the pen 5mm maybe  give yourself more room with $132. Can you play with the pendant and move the axis normally with that? (confirm that x moves x directions, y moves y directions, and z etc etc) 

if all that is good in the directions, then verify that the g-code is also doing what you expect.
otherwise sounds like may be an issue with electrical.

def670, 2018-04-13:
The switch does not trigger when just being lifted.  it homes out properly (z-axis first which DOES NOT fail), then x and y.  After that i have it do a job.  Any job.  from gear to writing - as soon as the job finishes the pen lifts, and continues lifting until it trips the z-axis limit.  It is visible - meaning i watch it physically hit the limit.  I have been through  at least 50 jobs where this has happened (continuiously).

I do not know about the pendant as i have not been using it.  in the early stages when i got into the pendant it operated as expected.  The limits were added in over the last week or two an have been a pain in my brain since.

def670, 2018-04-14:
ok issue solved.
I had too large a number of steps on my z-axis stepper motor (it has a gear fixed to it which is lifiting the pen, and this was not taken into account.  I do not have it fully figured out yet, but when i halfed the number of steps in $102 i stopped hitting the limit.  Epiphany as drifiting off to sleep.  Hallijuha

closed 2018-04-14 by def670