vlachoudis/bCNC #804: Zooming Viewport with the mousewheel only works in one direction

zirkeltraining, 2018-04-15:
Thank you for your software, it's great to get into CNC. 

A minor annoyance: 
on a Mac, OSX 10.11.6, using Python2, (the only config that i tested),
i cannot zoom OUT with the scroll wheel of my mouse (or two-finger-swipe-down). It only zooms into one direction, IN.  That's kind of frustrating because it's what every CAD tool uses, and navigating the viewport is clumsy without it.

vlachoudis, 2018-04-16:
I don't have a mac to test, but its strange since up to now I've got the impression that it was working on mac.

Harvie, 2018-10-29:
Works for me. Do you still have this problem with lastest master?

fedulvtubudul, 2022-02-20:
Issue is still here and it's really boring. I've tried to debug and it seems related to Python 2/3 rounding rules differences. Please check this PR: https://github.com/vlachoudis/bCNC/pull/1678. Works for me, but I've only tested on Mac.