vlachoudis/bCNC #805: Disable processed blocks

Guenni75, 2018-04-16:
Hi.
At large jobs, i have to make it at 2 or more days.
When i stop the job, i have to search the last finished block .

It would be great, if there where an option, to disable a block, when it is finished.
in this case, i only have to check, if it is not only send and i can save it for the next day.

buschhardt, 2018-04-16:
Hi, I think also save parts of gcode (and of course append/insert as parts) would also be nice.

Harvie, 2018-07-19:
I think we might use some feature called "start from selected block/line"

Guenni75, 2018-09-07:
I can start the Job at every Block, if i enable the Header. I never had Problems with that.
I don`t need commands from the previous Block.

It would be enough to disable a block after complete processing.
The Header must be enabled.

Harvie, 2018-09-07:
@Guenni75 This is not really a good idea. Imagine what will happen if you want to cut the same part again... You will need to reenable all the blocks, but some of them were "construction blocks" that were never meant as actual toolpaths. So you will have to think it through again to enable only the right blocks.

That's the reason why i don't think we should reuse the enable/disable block functionality for this.

Harvie, 2018-09-10:
@Guenni75 
> I never had "construction" blocks or something like this.

That's because you don't generate the code using bCNC, but rather use external CAM. That's no problem, but we can't break workflows of people (like me :-) using internal CAM.

But don't worry. We can certainly implement feature to resume operation. I think it's really useful idea!

Guenni75, 2018-09-11:
Great!

Harvie, 2018-09-11:
@Guenni75 So you want to use STOP button, then save the ngc file, open it another day and hit RESUME, right? I think we can just add some bookmark which will indicate last cut block and last cut line. Then we should add buttons to start g-code from selected line/block. And we should add resume button, which will start from the saved bookmark.

At least that's approach which would feel reasonable to me.

But there's one catch. I would really like to get this working in machine failure conditions. Eg.: so i can resume my cut after power loss. There are two possible levels of power loss:

1.) CNC machine lost power, but laptop is running from battery. This is relatively easy, but it's hard to detect, because arduino will be still powered via USB from the laptop battery, so it will respond via serial, but motors will not run. I guess some brownout detection circuitry can trigger arduino's "abort" pin.

2.) Both CNC machine and PC will lose power. This is tricky, because at that point you can't save anything. We can periodicaly (eg. every 15 seconds) write the progress to bCNC.ini and ensure it was flushed to disk, so we have at least some checkpoint after reboot.

BTW how do you plan to re-zero your work coordinates after powering it down mid-job? Some people use endmill to drill small hole in 0,0 coords to get physical reference point.

Guenni75, 2018-09-12:
@Harvie A Resume Button would be nice, but (for me) it would be enough to set a bookmark to the finished blocks, or just disable them. 
Your thought to resume after Power loss  goes much further.
I never had such problems, but you never know. Could be very useful.

After powering it down mid-job, i just do a home cycle and i can resume at avery block i want.
As long as I do not move the workpiece (two side machining for example), i never need a physical reference point.

nonpds, 2021-04-22:
I also want to be able to start my gcode at a particular line. USG allows a "Run from line", simply enter the line number of the gcode and it begins from that point.

massiveradiation, 2025-06-10:
> Hi guys, I'm using bCNC for a while and some things are unclear to me.
> When there's an alarm stop, or a pause, I find the line number at the
> bottom left. How can I find out which block that line corresponds to? How
> can I tell bCNC to start over from that line? Thanks for your help and
> sorry for my perhaps trivial requests.

Hello,
How I resume after a crash/alarm:
- Go to editor tab
- show and select the exact point in the tool path you'd like to resume
from in the preview window. This will highlight the block.
- Uncheck previous block "X" except header. A trick is to go to the left
panel and select the block above the highlighted one. Then hold down shift
and select the "X" box after the header block. This will uncheck all boxes
prior to your desired start point.
- If you're using an automatic tool changer, you'll want to swap in the
current tool in to your header code.
- Now when you hit run, only the checked blocks will execute. Be careful
not to save the file.

I use this process daily and it's fast and efficient.
Good luck!