vlachoudis/bCNC #807: scan function try to use empty coords if none of pathview type is selected at file loading

tuxun, 2018-04-17:
Hi, i load laser gcode heavies file in bcnc, and i sometime need to disable the rapid paths (g0) and "machine working" paths (g1) in view for being able to load such huge files.

Yesterday, i did it this way, and when i tried to use "scan" function to see if the gcode to engrave was fitting the material area, it "bugged": my machine start to move, i suppose to 0, then it told me something about scan probing wasnt working with null or badly set coordonate (+1 for the precision in the message).

Can someone double check that if we load gcode without showing paths, margin arent computed but we are able to run/send to controller a scan function (so maybe the message was more like "unable to scan 0,0,0,0 cause last coords arent positive") i use the master dled one weeks ago :)

-also a message confirming that this post is understandable, i maybe let one mistake or two as english isnt my native langage
PS: i continue diging and may come back with a "how to reproduce" procedure and a copypaste of that message :)

vlachoudis, 2018-06-04:
The scan feature calculates the boundaries of the gcode when it is plotted (unfortunately).

tuxun, 2018-06-04:
if someone have an idea of how to get boundaries of gcode loaded but not rendered, I would be happy to hear it :)