vlachoudis/bCNC #812: Question: Camera to align double face PCB?

sle118, 2018-04-22:
I was searching through the documentation on how to use the camera and stumbled on the following issue:
https://github.com/vlachoudis/bCNC/issues/140

This seems to have been the starting point of the camera alignment feature, which I have used previously. Reading through the conversation there, though, it seems that this may be the best to around to align a PCB   to 2nd side milling! 

I'm thinking about the following procedure: 
Attach blank PCB to machine, on a good squared reference point
Run auto level
Drill holes
Mill traces
Turn around
*** Align known points from board using camera??
Run auto leveling
Mill 2nd side of the board

Would this work, and  has anyone done this before? It would be nice to write a wiki on the topic.  I might volunteer to do so.

buschhardt, 2018-04-22:
I think before "turn around" two reference points (holes) - with a maximum distance would be good, because after turn its possible that you twist the z-axis of the pcb a little bit.

PS: I only do one side pcb to now.

sle118, 2018-04-22:
I didn't think of twisted z axis, but I guess at these tolerances, it's very likely that even the slight angle could cause issues. I've been milling single sided boards too so far, but I feel that they are rather limited and need some extra effort to design.

Thank you for the feedback!

closed 2018-06-20 by sle118

vlachoudis, 2018-06-22:
@sle118 you are welcome!