vlachoudis/bCNC #816: Feature request for external Feed Rate Control

einencool, 2018-04-27:
Hello, I would like to ask, if it would be possible to add the feature to control the feed rate with a „pots“? (Is this really the word? In German it is „Potentiometer“)

With the Software „Estlcam“ there is a feature on the Arduino Mega on the Pin ADC 1 and 2 to control the feed rate and the spindle speed. Because I love your Software, it would be great if you could add this features.
In „Estlcam“ there is altough the feature to jog the Spindle to a desired place with a „hand wheel pulse encoder“ and the Arduino Mega.
This would be altough a great feature for future Versions of your wonderful program.

Greets Chris

buschhardt, 2018-04-30:
Hallo Chris
I think first we should find out how bCNC get the values from the variable resistor to process this as variable?
I see bCNC has scripting posibilities to inject values into GRBL (wiki). 
So I would use an extern server that handle the converting from ADC to usefull values. 
Grüße Thomas

Harvie, 2018-10-29:
This will happen only if you convince GRBL people to provide analog readings from ADC in status reports sent over serial. Which is not very likely to be done on arduino platform due to lack of memory. You might get lucky with 32b port of GRBL that is in development.

closed 2018-10-29 by Harvie