vlachoudis/bCNC #819: bCNC stops and I have to restart...

enrico1975, 2018-05-01:
Hi, I'm using bCNC with a Cyclone PCB Factory with GRBL... everytime I send a gcode it stops without errors and I have to close all and restart.
The gcode is:
``G21
G90
G00 Z5.0000
M03
G4 P1
G00 X21.8415 Y15.8725 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X21.8415 Y13.3731 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X16.1265 Y5.0775 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X7.2365 Y10.4750 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X25.1663 Y9.2050 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X7.2365 Y13.0150 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X25.1663 Y4.7041 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X5.9665 Y2.5375 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X29.7790 Y1.9025 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X14.8565 Y7.9350 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X7.2365 Y15.5550 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X5.9665 Y5.0775 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X14.8565 Y10.4750 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X18.6665 Y9.2050 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X18.6665 Y4.7041 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X14.8565 Y13.0150 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X14.8565 Y15.5550 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X19.6190 Y1.9025 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X16.1265 Y2.5375 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X7.2365 Y7.9350 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X30.7315 Y14.9200 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X35.4940 Y11.1100 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X2.4740 Y5.3950 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X35.4940 Y13.6500 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X2.4740 Y7.9350 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X35.4940 Y16.1900 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X2.4740 Y10.4750 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X2.4740 Y13.0150 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X30.7315 Y12.3800 F80.00
G01 Z-2.0000
G00 Z5.0000
G00 X0.0000 Y0.0000 F80.00
M05
``

vlachoudis, 2018-05-10:
Something is dropping the connection. Electrical noise maybe?

MildaGenius, 2018-07-05:
I dont know if is  the same problem, but sometimes in middle of milling process motors stops, spindle switch off, but bCNC time in bottom bar still running. No additional error. bCNC never resume milling :(.

Its really anoing, that process stop in middle of task.

I must reconnect to CNC and restart milling.

[image]

[Batman_spinner_290_40_4.8.log](https://github.com/vlachoudis/bCNC/files/2167613/Batman_spinner_290_40_4.8.log)

OrangePiPc + Armbian + bCNC 0.9.14 (today fetch)

Harvie, 2018-10-30:
Most likely the HW problem. What spindle do you have? Do you have shielded cables to motors and spindle? How long USB cable? do you run the cables along each other?

Please experiment with this and let us know if that helped

MildaGenius, 2018-11-16:
To motors and spindle I have shielded cables. But in controller case I have unshielded part (cca 200mm) to connectors.

When I disconnected all end switches and e-stop from CNC shield, the problem was gone. I have prepared PCB with optocouplers to filter signals from end switch, but I didnt have time to test it yet.

Iam preparing control case upgrade with shielded cables to connectors and drivers change.

I can confirm, that in my case was HW problem.

Harvie, 2018-11-20:
@enrico1975 Was that the same case of HW problem for you? I am closing the ticket. Let us know if this was not HW related.

closed 2018-11-20 by Harvie