vlachoudis/bCNC #820: Do not go up at each level in the Pocket operation

MARIOBASZ, 2018-05-03:
Hi, i thins It's a good option that does not go up to safe z to go back down in each level of milling in the operation of pocket.
(Sorry for my english)

vlachoudis, 2018-05-04:
Thanks for the remark. The pocketing algorithm is very primitive. It goes to safe z-height just in case that it might intercept a part which is outside the pocket. Clearly the algorithm there needs that the fast motion doesn't intercept anything. I am closing the issue here, since this goes together with the improvement of the pocket and the differences

closed 2018-05-04 by vlachoudis

MARIOBASZ, 2018-05-04:
Hi, Sorry to express me wrong. The change is in the cut operation, when the final position Is differentt from the initial position