vlachoudis/bCNC #821: Helical cuttig

MARIOBASZ, 2018-05-03:
Hi, despite my programming limitations, this is my first add-on. When helically cutting to the chosen depth, it allows drilling larger holes than the milling cutter and with the right tool to create threads. 

[image]

[Helical_Descent.zip](https://github.com/vlachoudis/bCNC/files/1972902/Helical_Descent.zip)

vlachoudis, 2018-05-04:
Many thanks @MARIOBASZ I will give it a look and integrate it in the program

vlachoudis, 2018-05-04:
I've just added it in the master

closed 2018-05-04 by vlachoudis

Harvie, 2018-05-06:
Would it be possible to center helical cut on selected object (like "drill" does) instead of entering coords manualy? Manual coordinate entry is usefull, but i guess it would make sense to center at selected object by default and only enable manual coordinate entry by checkbox.

Harvie, 2018-05-06:
Also for thread milling you don't always want the flat circle on top and bottom. Eg.:

[image]

On the left is thread milling toolpath created by this online tool https://sct-usa.com/thread-mill-code-generator/
On the right is "right thread" operation from helical plugin in bcnc

I guess there should be checkbox to enable/disable these... In case you are cutting threads it just removes material that is probably not needed to be removed. In case you are cutting hole it does not really make sense to cut first at the Z 0, so the flat circle on top rarely makes sense. Even cut operation has "cut first at surface height" checkbox.

Harvie, 2018-05-07:
Also just realized, that helical plugin currently does only negative tool radius offset (for cutting internal diameters). It would be vital to also have option to cut with positive offset (for OD) to simplify calculations when cutting threads. Also disabling offset completely (toolpath on line) can make sense in some scenarios. Please add that also.

BTW i just found this on youtube. Looks like dead simple and cheap DIY tool for thread milling:

[image]

MARIOBASZ, 2018-05-09:
Hello Harvie, I'm glad that my plugin is useful for you. From your observations, if I interpret well:
1-I will upload the modification or another version so that the perforation is made at the beginning of the chosen Block or a series of helical perforations.
2-I do not understand your observation, there are two helical modes in Helycal Type: cut (Helical cut), or create threads (Right / Left Thread). In the second case it does not make a flat cut neither in the first nor the last level (let me know if I'm wrong). The initial height is selectable
3- You are right, I must add the tool exit option outward for external thread.
I mention that to make threads in engineering plastics, I use end mills that are used in jewelry for crimping. A flat end mill can be used, but only if the thickness of the wire matches the height of the latter; would have to modify the code to make it more versatile.
Excuse my bad English, greetings, Mario

MARIOBASZ, 2018-05-09:
While I was taking a piece of pizza out of the oven, it occurred to me that it is not as complicated to make threads of thick thread as those of wood in your photos. Until you implement it, manually, you can use the cut command to descend in different passes, shifting the thread

Harvie, 2018-05-09:
> here are two helical modes in Helycal Type: cut (Helical cut), or create threads (Right / Left Thread). In the second case it does not make a flat cut neither in the first nor the last level (let me know if I'm wrong). 

For me it does flat paths every time, no matter what i set. I guess this would make more sense to set these independently. That way you'll be able to also do left handed helical cut or cut without top, but with bottom.

> You are right, I must add the tool exit option outward for external thread.

It's not only about tool exit. It's also about how you compute the diameter of helix. right now you subtract radius of tool to calculate helix diameter. (it's same as profile operation inside vs. outside)

> you can use the cut command to descend in different passes, shifting the thread

you can't apply "cut" operation on helix, since it only works with 2D shapes.