vlachoudis/bCNC #822: Problem with Profile in Command Line in the direction inside

MARIOBASZ, 2018-05-04:
On the command line, if I write:
profile 1 in
make the route out for that tool value!
Only allows the internal route for the tool loaded in the software.
Another point: It would be nice to add an offset as another parameter, either in the form:
profile 1 0.5 out
or
profile 1 out 0.5
or
profile 1 + 0.5 out

vlachoudis, 2018-05-04:
Try putting a sign on the offset
```profile 1```    (outside)
```profile -1```  (inside)
or
```profile i[nside]```
```profile o[utside]```
using the default tool

MARIOBASZ, 2018-05-04:
Thank you! Although it works in the same way in / out as -1/1, it does what I want, it writes the diameter of the tool with its corresponding sign

closed 2018-05-10 by vlachoudis