vlachoudis/bCNC #824: Error reporting endpoint seems to be offline

Harvie, 2018-05-06:
When something fails i get this error reporting window:

[image]

But when i click on "send report" it only gets stuck until it eventualy times out showing this:

[image]

I guess that the website with error reporting api is offline, which makes the error reporting window just another annoyance. First you get annoyed by bcnc failing, which shows you automatic error reporting window, which in turn annoys you even more...

buschhardt, 2018-05-09:
Yes the fixed address is not reachable: 
http://www.bcnc.org/flair/send_email_bcnc.php

@vlachoudis  When I can help You, drop a mail. :-)

vlachoudis, 2018-05-10:
Indeed the address is not working. I am looking for an alternative. Does anyone know a free hosting site (maybe google sites) that could run such a script?

Harvie, 2018-11-20:
Do we still need such feature? I crash bCNC a lot and it would create tons of error reports. We have barely time to respond to issues here on github. I can't really see someone watching automated reports on this scale. I know it can be automaticaly preprocessed to remove duplicates and legacy versions. But still... I will leave this for discussion.

Harvie, 2018-12-14:
Someone reccomended me to take a look at https://sentry.io/
It might be usefull in this case. Or not. We have to test it.