vlachoudis/bCNC #825: Show commands that were initiated by clicking on buttons

Harvie, 2018-05-06:
#822 inspired me with this one. When i click on some command like "profile". It would make sense for bcnc to show the executed command somewhere at the bottom of the screen in status bar and somewhere in command history. That would make it easier for newbies to learn the commands.

Also having way to issue these commands from commandline when launching bcnc might be cool. That would provide some basic scripting abilities... Like: load dxf, profile the shape, export dxf, cut profile, export gcode and exit purely from commandline without even showing GUI window.

MARIOBASZ, 2018-05-10:
When executing, below to the left appears the leyeda. In the case of profile, writing on the command line "Profile block distance = xx" appears; with "Generate profile path" button

Harvie, 2018-07-01:
Just figured out, that all of this is already done for buttons using `insertCommand("SOMETHING", True)`. But not every button does this (eg.: profile).