vlachoudis/bCNC #828: Second version Helical plugin

MARIOBASZ, 2018-05-10:
I have made some improvements in my Helical plugin. 
Performance: Milling / threading right / left, internal / external.
Choose diameter and pitch of the helicoid.
Choose the end mill (or use the last one selected if not chosen)
Choose the starting location of the helix from the panel or from where the selected block starts (it would be better than in this option, the helicoid is added at the beginning in the same one instead of in a new one.)
 My knowledge of Python still they are insufficient, I have based on Filippo Rivato's Driller plugin and Buschhardt's improvement, if someone wants to implement it, thank you, or new option to include in the original block / generate new Block).
Choose the starting and ending heights, which allows milling / tapping away from the end of the material.
Set the safety margin from the edge at the input and output of the tool.
Milling / tapping at a height higher than the initial one (I do not know if it has any use).
Eliminated passed in initial and final levels for threading, only remains in type of helical cut.

Interesting improvement if someone wants to add it: To make a thread of fillets wider than the height of the Bit type T, it would be necessary to generate copies displaced in height for each successive pass.
Happy with the result, tell me if there are errors and I hope it is useful.
Thanks again Viachoudis for your software, to those who collaborate, especially to Rivatto and Bruschhardt for your work and Harvie
for your observations (see what you have managed to achieve!)
Greetings, Mario (from Argentina)
[Helical_Descent (2).zip](https://github.com/vlachoudis/bCNC/files/1992134/Helical_Descent.2.zip)

MARIOBASZ, 2018-05-10:
Silly observation: for my ignorance of Github, should I have included it in the extraction request?

buschhardt, 2018-05-11:
Hallo Mario,
a nice plugin. It would be nice if you can make pull request, so we can see easier your changes.
Maybe this help you https://www.gun.io/blog/how-to-github-fork-branch-and-pull-request 

Thank you

buschhardt, 2018-05-11:
There occur a error:
```
Traceback (most recent call last):
  File "/home/thomas/Dokumente/fraese/bCNC/Utils.py", line 417, in __call__
    return self.func(*args)
  File "/home/thomas/Dokumente/fraese/bCNC/ToolsPage.py", line 1490, in execute
    self.tools.getActive().execute(self.app)
  File "/home/thomas/Dokumente/fraese/bCNC/plugins/Helical_Descent.py", line 229, in execute
    elif clearanceEntry < 0 or ClearanceEntry == "":
NameError: global name 'ClearanceEntry' is not defined
```
Here my settings:
[image]

MARIOBASZ, 2018-05-11:
Maybe you uploaded the version before it came back up with corrections, on my computer it generates the helix.
Uploaded in pull request

MARIOBASZ, 2018-05-11:
You're right, I corrected it and uploaded the modification to the pull request

closed 2018-10-30 by Harvie