vlachoudis/bCNC #832: Driller Anchor in first hole repeats and error Z

MARIOBASZ, 2018-05-11:
The first hole is done at the wrong height (random?) (Can you tell me why I learn so?) Then do it correctly and repeat it one last time if it is a closed route

[image]

It is a triangle, there should be 3 holes, there are 5 and the first is at Z = 6, when it should be z = -1
I add a pull request

buschhardt, 2018-05-12:
See https://github.com/vlachoudis/bCNC/pull/833 maybe it helps with closed path.

MARIOBASZ, 2018-05-12:
I found this error: if there is more than one block and you choose one different from the first one, it generates a first additional hole (I think there is a function to eliminate the points external to the route)

MARIOBASZ, 2018-05-12:
Can you guide me where to find list (OrderedDict.fromkeys (bidHoles) to understand how it works to clear duplicates?

MARIOBASZ, 2018-05-12:
A minor improvement could be to eliminate duplicates in xy plane (it does not make sense to make more than one hole if the movement is just change of height without translation)

buschhardt, 2018-05-12:
> Can you guide me where to find list (OrderedDict.fromkeys (bidHoles) to understand how it works to clear duplicates?

Yes, the OrderedDict(ionary)-Class is a special Hash that have also a reference to an array. In the normal case Dictionaries are unordered - you looking only for the key and get the value - it dosnt matter where the key is. But if you need the original order, then you can use this class.
How it works here:
1. take all listmembers as keys of this class [p1,p2,p3...]->{p1:null,p2:null,p3:null}
2. if the import occur there is only one key with the same name possible [p1,p1,p2]->{p1:null,p2:null}
3. fromkeys -> we interest only in keys
4. make a new list from that keys
Keys can also be other objects (without Dictionaries, I believe)
```
from collections import OrderedDict
bidHoles = list(OrderedDict.fromkeys(bidHoles))
```

> A minor improvement could be to eliminate duplicates in the xy plane (it does not make sense to make more than one hole if the movement is only height change without moving)

If You use the driller plugin - I eleminate only the duplicates from a block - not from all blocks. Should we?

MARIOBASZ, 2018-05-12:
Thanks
Suppose the previous block ends at a low height, Houstn we have a problem!
you just punched the neighbors below!

independently that the previous block is unmarked
GCode in .TXT

[PRUEBA ANCHOR DRILL.TXT](https://github.com/vlachoudis/bCNC/files/1997660/PRUEBA.ANCHOR.DRILL.TXT)

[image]

buschhardt, 2018-05-12:
I cant find an error, please describe every step to reproduce Your procedure. The textfile inst that from the screenshot.

buschhardt, 2018-05-12:
OK I see if I have 2 blocks selected and run the plugin, it will add the resulted gcode in one block.
But this is not an error, or?

MARIOBASZ, 2018-05-18:
I do not know why it is comparing with version 0.0.8 instead of 0.0.10
Below I clarified the few modified lines

MARIOBASZ, 2018-05-19:
I have seen where the problem, in anchor, continues to appear in the segments originated with G0 or Zonli