vlachoudis/bCNC #836: My translation is not loading

hordri, 2018-05-16:
I have started to make a translation for the program, but i cant get to load it. 
I have followed the instructions on the wiki, for the language of the translation i have selected magyar (Magyarország), which is set in my windows. I have checked for the locale codes and its correct in the name path, but tried it with all caps too, still no luck.
The files are located here: 
C:\Users\USER\Downloads\bCNC-master\bCNC-master\locale\hu\LC_MESSAGES

SteveMoto, 2018-06-19:
[hu] as language is not yet implemented in bCNC.

If you still want to test and use your translation, you can overwrite an existing language in bCNC (eg "de"). The hu.po and hu.mo must then be renamed de.po and de.mo and saved in the folder "de". In the program "Deutsch" has to be selected as language.

When the translation is done, Vlad can integrate the files (which must be called hu.po and hu.mo again correctly) into bCNC including language selection.

Greetz Steve.

vlachoudis, 2018-06-19:
(In linux) If you set the language to "System" and you start the program as
```LC_ALL=hu_HU bCNC```
it should load your translation

SteveMoto, 2018-06-19:
I read / think that hordri uses Windows. Could that be because of it, but can not test it myself, since I use only Linux here.