vlachoudis/bCNC #839: Wpos not updating on display

stewmac570, 2018-05-26:
Hi I have just started out using BCNC on my open builds mini mill. I'm using a smoothieboard and I'm finding the the Wpos is not updating. It is setting the G54 (or G55...) so the machine is working in the right space but graphically the display and tool path is not correct. 

I have used BCNC on windows 10 and on Pi with the same issue so i think it maybe a smoothie setting. 

Can anyone shed some light on this?

stewmac570, 2018-05-27:
[image]

stewmac570, 2018-05-27:
this is where i am up to. you can see the bottom left "Set workspace G54 to X0" but the WPos display is showing the MPos value. then in the tool path display the red arrow representing the tool tip (?) is out side the work area. 

I'm sure there is some work i need to do with the + and - of the display and the USC orientation on the CNC file. But the WPos is my biggest issue. 

when homing the machine the red arrow goes to the right rear and displays that correctly. but should it not change when i set the Wpos? 

Any help on this matter is greatly appreciated.

stewmac570, 2018-05-28:
Its ok. i found the issue. from the connection tab there is a drop down box and you need to select smoothie

vlachoudis, 2018-05-28:
Ok, Thanks for reporting

closed 2018-05-28 by vlachoudis