vlachoudis/bCNC #840: Add overcut possibility for pockets

Harvie, 2018-05-28:
When i want to make pocket with overcuts, i have to first create pocket operation without overcuts and then add inside profile operation with overcuts, so outline of pocket is cut again (mostly cutting air) with overcuts added.

I think it would be cool to have overcuts generated directly in pocket generation, so i can save one pass and one CAM step when doing something like this:

[image]

MARIOBASZ, 2018-06-06:
A quick way is to write "poc" or "pock" on the command line after doing the profile with overcut, because the tool used is loaded

MARIOBASZ, 2018-06-07:
I agree that it is not ideal, you must do two steps. but it does not cut in the air: 1- you make the profile
2- Pocket operation runs on the resulting profile trace, does not cut over

Harvie, 2018-06-07:
I see, that makes sense. I've been doing the pocket on the original shape rather than on the resulting profile. That's way to remove the extra pass. However i think this should not be so hard to add to pocket operation, since all code is already there.

Maybe we can get the opposite approach and add "pocket" checkbox to profile operation, so that profile will be automaticaly pocketed creating single compact operation. That way it would no longer make sense to have separate pocket operation as it would be the same as inside pocketed profile without overcuts...

Harvie, 2018-06-25:
Also it does not work in some cases. I've tried this on my drawing and the inside profile generates OK, but the pocket of that profile generates outside of it:

[image]

Harvie, 2018-06-25:
BTW i've found way to profit from this. I first do the pocket and then profile with overcuts in reverse direction, so it works like spring pass for the pocket. However i would still rather be able to choose wheter i want the extra pass or not. :-)

Harvie, 2018-06-30:
I've added little checkbox to create pocket and profile in single step:

[image]

Harvie, 2018-10-30:
Somewhat implemented. I will rewrite pockets completely soon, so it will get even better.

closed 2018-10-30 by Harvie