vlachoudis/bCNC #841: Displaying curves error

neomadic, 2018-05-29:
Sorry if this has been covered.  I tried searching here and can't find anything.

Is there something I can do about curves being displayed as straight lines?  I've tried changing plotting arc accuracy up and down but it doesn't help.  It seems like the tool path is running correctly though.

Here are some screen shots:
[image]
[image]

vlachoudis, 2018-05-30:
@neomadic could you send your gcode to check. All curves/arcs are always converted to line segments for plotting, there are two limitations one is in the maximum acceptable sagitta and the second on the length of the segment.

neomadic, 2018-05-30:
I can post it up this afternoon.  I went ahead and cut the knife out last night and the arcs cut perfectly so this is definitely just a display plotting issue.  It's really not a big deal to me at all now that I know everything cuts right.

Harvie, 2018-06-12:
What is the program used at the first screenshot?

neomadic, 2018-06-12:
Sheetcam.

Harvie, 2018-06-12:
@neomadic thanks. Anyway... see #587 it's exactly the same problem that i've reported some time ago and there's even mentioned how to fix that by changing configuration... I think this should be solved by default, but i guess then bCNC would have problems with performance. So it will probably not get real solution until refactoring bCNC to use OpenGL...

neomadic, 2018-06-12:
I'm usually a 100% FOSS guy but I'm really loving Sheetcam so far.  

I can see how perfectly plotting every arc in the display would be pretty resource heavy.  So far, I've only had the issue with one file.  I've done enough playing with gcode lately to reawaken the part of my brain that used to be good with it so I'm much more confident in my ability to check code now.  The bCNC display is good for a quick visual check to make sure I have the right file loaded.  As far as I'm concerned, it doesn't need to be perfect.

I'm going to go ahead and close this since it's a known issue and doesn't affect performance.

closed 2018-06-12 by neomadic

vlachoudis, 2018-06-12:
@neomadic it is not the tkinter vs opengl that will increase performance. The bottleneck is using python to reconstruct the paths from the gcode, and this part will be the same, unless if it is re-written in C, but then it wont be portable across various OS without recompiling.

neomadic, 2018-06-12:
That's all still way over my head at this point but I'm sure I'll gain enough knowledge to contribute soon if I stay on my current path.   The last time I used C was in 1993, and I don't remember it being a fun time.