vlachoudis/bCNC #844: bCNC TinyG and Pi3

Seb66, 2018-05-31:
Hi,

I have changed out my CNC controller to a TinyG and cannot get the USB port/Connection active.
There is no USB driver other than usb0 and AMAO. When i open usb 0 the prompt says it connected but doesnt change the orange button?

How do i get tinyG to work with bCNC on a Pi3?

vlachoudis, 2018-06-04:
I have a version which **works** (a big word) with TinyG. It lacks still many features in order to be compatible.

MARIOBASZ, 2018-06-04:
Excellent! I want to test the operation of the machine with jerk control and so far only G2Core / Tinyg has implemented it. It would be nice to be able to handle it from BCNC

colinluthier, 2018-06-17:
@vlachoudis Is the TinyG version available as a Beta?

Harvie, 2018-12-01:
Duplicate of #83

closed 2018-12-01 by Harvie