vlachoudis/bCNC #847: Workspace issue

DaveSkowron, 2018-06-04:
I load a g-code file generated by DXF2GCODE and the workspace is so far away from the actual work that my machine errors out due to limits before it even gets close to the work. The workspace centers on the drawing in DXF2GCODE. Why is there such a huge offset?

vlachoudis, 2018-06-06:
metric vs imperial? Maybe some the settings of bCNC are not compatible with grbl and dxf2gcode?
You don't need dxf2gcode, since bCNC can load directly dxf.

DaveSkowron, 2018-06-06:
Loading the DXF file directly into bCNC renders the part tiny - like 1/10th scale. I found out how to move the part into the workspace but cannot find a way to scale it properly. I'll go back to the beginning and start over. Maybe I did something wrong somewhere along the way.