vlachoudis/bCNC #856: Feed rate control with shortcuts

einencool, 2018-06-13:
Hello, 
I‘m running bcnc on a Arduino Mega with Grbl 1.1
In the Software I can change the feed Rates with the slider when the program is running.
I have a external num Pad and would like to set the codes to change the feed rates +10% and -10% to two shortcuts so I’m able to route them to my key pad.
Can you help me, what I have to insert for the Shortcuts?

Many thanks.
Greets Chris

colinluthier, 2018-06-17:
I am looking into this as well. The commands in 1.1 are pretty simple but involved enhanced ascii. I am not sure how Python handles them. I was trying o make shout cut buttons for some of these:

"0x85 : Jog Cancel

Immediately cancels the current jog state by a feed hold and automatically flushing any remaining jog commands in the buffer.
Command is ignored, if not in a JOG state or if jog cancel is already invoked and in-process.
Grbl will return to the IDLE state or the DOOR state, if the safety door was detected as ajar during the cancel.
Feed Overrides

Immediately alters the feed override value. An active feed motion is altered within tens of milliseconds.
Does not alter rapid rates, which include G0, G28, and G30, or jog motions.
Feed override value can not be 10% or greater than 200%.
If feed override value does not change, the command is ignored.
Feed override range and increments may be changed in config.h.
The commands are:
0x90 : Set 100% of programmed rate.
0x91 : Increase 10%
0x92 : Decrease 10%
0x93 : Increase 1%
0x94 : Decrease 1%
Rapid Overrides

Immediately alters the rapid override value. An active rapid motion is altered within tens of milliseconds.
Only effects rapid motions, which include G0, G28, and G30.
If rapid override value does not change, the command is ignored.
Rapid override set values may be changed in config.h.
The commands are:
0x95 : Set to 100% full rapid rate.
0x96 : Set to 50% of rapid rate.
0x97 : Set to 25% of rapid rate."

colinluthier, 2018-06-17:
Reading through the code I found the slider. Doh! Thank you for being one step ahead!

einencool, 2018-06-18:
Hi,
Did you only enter the 0x85?
I got the hint to restart the software to take the changes effect. But I didn't tested it yesterday. I hope I'll give it a try today.

einencool, 2018-06-24:
I tried it today, but I only get error messages.
When I enter 0x90 or something similar, the message says unknown command
When I enter the sign with "alt + 144" there comes a sign I can't read, but there also comes an error.

Could you please help me, how to enter this code? I would like to enter them trough the shortcuts "Shift + F7" - "shift + F9" 
On this keys I've mapped the keys, and they worked for run / pause / Reset very well.

Thank you in advance

einencool, 2018-06-24:
Hello, 
Now I tested it, with the parameter "sendhex 21" and it gave me the ability to hold the machine, but when I enter "sendhex 144" for the speed adjustment, it doesn't change the speed.
[https://github.com/gnea/grbl/issues/477](url)

This is from the python log, I don't know if anybody will help this:
Traceback (most recent call last):
  File "D:\bCNC\Utils.py", line 417, in __call__
    return self.func(*args)
  File "D:\bCNC\bCNC.py", line 578, in <lambda>
    self.bind("<%s>"%(key), lambda e,s=self,c=value : s.execute(c))
  File "D:\bCNC\bCNC.py", line 1645, in execute
    rc = self.executeCommand(oline)
  File "D:\bCNC\Sender.py", line 388, in executeCommand
    self.sendHex(line[1])
  File "D:\bCNC\Sender.py", line 606, in sendHex
    self.serial.write(chr(int(hexcode,16)))
ValueError: chr() arg not in range(256)
(<type 'exceptions.NameError'>, NameError("global name 'self' is not defined",), <traceback object at 0x000000000B91D8C8>)

closed 2018-06-24 by einencool