vlachoudis/bCNC #857: How to Change Y-axis direction during homing sequence

jaredc23, 2018-06-17:
Currently, I am running a homing sequence with the settings set the the top left, which means x axis all the way to the left, bed up at the front, and spindle all the way up. Is there a way to make the Y-axis go backward instead of forwards, but keeping X-axis and Z-axis the same?

toktuff, 2018-06-17:
This isn't a bCNC issue but something you configire in your CNC-controller. If you are using grbl you can change this with setting $23. I think you want to set it to to 2. But you'll have to verify this.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#23---homing-dir-invert-mask

jaredc23, 2018-06-17:
Thank you for responding so quickly. To invert the Yaxis only, so x moves left, Y moves back, and Z moves up, it is $23=3. $23=2 inverts X and Y

closed 2018-06-17 by jaredc23