vlachoudis/bCNC #875: bCNC mirror image

benek1, 2018-06-22:
Hello. I built a Cyclone pcb factory, the machine is controlled by bCNC. I noticed that generate a string in the program, it is displayed correctly but milled inversely (in a mirror reflection) is it normal, am I doing something wrong? I will add that the g-codes generated in other programs, e.g. copprecam, I also have to bounce to be milled correctly.
If there is someone who knows something about it, please help
PS sorry for English.

MARIOBASZ, 2018-06-22:
Hello, it occurs to me that the x axis is running inverted

vlachoudis, 2018-06-22:
@benek1 wrong wiring on the motor, or wrong configuration of the motor drivers. It should have nothing to do with bcnc. I am closing the issue, if you believe it is a software bug by opening a terminal and manually sending the commands and goes in the right direction then please reopen it

closed 2018-06-22 by vlachoudis

benek1, 2018-06-22:
I got a hint of an email. The problem lay on the side of the direction of rotation of the engines. Grbl enabled the appropriate setting. Thanks for the help