vlachoudis/bCNC #876: Move spindle to camera location

Harvie, 2018-06-25:
It's nice to have camera alignment feature. But i really miss button to set zero to camera location, so i can find spot on stock and then move spindle directly over it. Right now when i zero the coordinates it is zeroed to where spindle is. Not where camera is. I want button that will set WPOS to negative camera offset, which means that spindle home will be exactly where the camera is now.

neomadic, 2018-07-02:
I haven't got too deep into this but I thought there was a camera to tool offset setting somewhere.  I think you make a mark at the camera crosshair point, drive the spindle over to it, noting the starting and ending positions, and enter the offsets in somewhere.  It would also be simple to drive the spindle to visual zero with a simple gcode command once you know the offset.

neomadic, 2018-07-02:
I thought that the offset settings in the camera setup automatically accounted for that?  If it doesn't, I'm not a coder but it doesn't seem like that would be too hard of a feature to add.

Harvie, 2018-07-02:
It can't just account for it. Because you would loose way to zero wpos on spindle :-) you need two buttons, so you can decide which one you want zero.
And Yep. That should be easy to implement, but i don't have camera on my CNC yet. :-)

neomadic, 2018-07-02:
What would be the reason you'd want to zero from the spindle instead of the camera?  I'm just curious.  My neck hurts so once I go camera, I can't imagine a scenario where the hunch over and squint zeroing method would be necessary.

steppel, 2019-04-29:
Uhm, having the same confusion right now. Where can I sent the spindle to the camera position. Right now I take the offset calculated in the camera module and then calculate the value and write the line to drill the hole in the pcb, that’s a pain. Where exactly do I find that button or option? I didn’t find anything in the documentation beside the camera calibration, it doesn’t explain how to utilize the offset values. 

Any help appreciated!

Harvie, 2019-04-29:
[image]

MARIOBASZ, 2019-04-30:
In the camera values, in addition to X and Y, there is Z: this height is the one that will move from camera to spindle and from spindle to camera: check that that height is suitable for the spindle (or the lowest)

steppel, 2019-04-30:
Awesome thanks guys! Totally not recognized the button as such.

Can I add the button to the Control tab?

Harvie, 2019-04-30:
I guess it might be possible to implement this as command and that would make it possible to add it as custom button.

steppel, 2019-05-01:
Any hint on how to turn that into a command?

Harvie, 2019-05-01:
just found there's already such command:

```python
                # CAM*ERA: camera actions
                elif rexx.abbrev("CAMERA",cmd,3):
                        # FIXME will make crazy the button state
                        if rexx.abbrev("SWITCH",line[1].upper(),1):
                                Page.groups["Probe:Camera"].switchCamera()

                        elif rexx.abbrev("SPINDLE",line[1].upper(),2):
                                Page.frames["Probe:Camera"].registerSpindle()

                        elif rexx.abbrev("CAMERA",line[1].upper(),1):
                                Page.frames["Probe:Camera"].registerCamera()
```

Commands are CAMERA SWITCH, CAMERA CAMERA, CAMERA SPINDLE, you can put it to the button...

steppel, 2019-05-01:
OMG that’s awesome. Thanks much!

MARIOBASZ, 2019-05-02:
Excellent, I had not seen it! Maybe it could be used to align another tool, such as a welding tool dispenser.
Can the button be copied as it is to the control tab?

Harvie, 2019-05-02:
[image]