vlachoudis/bCNC #877: Import button not easy to find

Harvie, 2018-06-25:
I've been looking for import button here in file menu:

[image]

For several days i was thinking that the feature was removed, but then i've found it's just hidden in editor menu:

[image]

Harvie, 2018-06-30:
I've tried adding button to `FilePage.py` and then in `bCNC.py` add this:

```python
                self.bind('<<Import>>',           self.importFile)
```

It gives following error, don't know why:

```
Traceback (most recent call last):
  File "/home/harvie/Temp/bCNC/Utils.py", line 421, in __call__
    return self.func(*args)
  File "./bCNC.py", line 2005, in importFile
    fn,ext = os.path.splitext(filename)
  File "/usr/lib/python2.7/posixpath.py", line 98, in splitext
    return genericpath._splitext(p, sep, altsep, extsep)
  File "/usr/lib/python2.7/genericpath.py", line 99, in _splitext
    sepIndex = p.rfind(sep)
AttributeError: Event instance has no attribute 'rfind'
```

vlachoudis, 2018-07-01:
You should call importFile without any argument like 
```lambda x,s=self: s.importFile()```

Harvie, 2018-07-01:
[image]

coming in next pull request :-)

closed 2018-07-02 by vlachoudis

avilab84, 2018-07-15:
The new button is not usefull !

Harvie, 2018-07-15:
@avilab84 it's not new button, it's old button in new place. also if it's not usefull for you then simply don't use it. problem solved :-) what works for one, might not work for other. i don't use all bCNC buttons either.

commit 1c8f87d9 (2018-07-02): Import button on file page, fixes #877