vlachoudis/bCNC #885: Start cycle hardware button

einencool, 2018-06-28:
Hello,

I‘ve got a question. I‘ve added some hardware buttons to my CNC using GRBL 1.1 with a Arduino Mega.
With these buttons I‘d like to „Start“ / „Pause“ and „Reset“ my Jobs.

The Pause and Reset buttons work well, but I can‘t start the job with the „cycle_start“ command.

Can you please help me, how I can change this? Is it a Problem with GRBL or can I code it anywhere?

Thank you in advance
Greets Chris

Harvie, 2018-06-29:
Probably related to #773

Lester444, 2018-07-18:
@einencool,

I wanted the same feature but now I use F12 for run.

Edit in 
       Tools ---> shortcuts   (add the word "run")

einencool, 2018-07-18:
I do the same, but I've got the option to wire the start pause and reset button directly to the Arduino. The reset and Pause Buttons work great, but the start button only works, when the program is running and then it's halted. Otherwise the start button is without function.

I've set a usb num pad with 12 shortcuts. But when I have the option to wire it directly, I would choose it...

Harvie, 2018-07-18:
@Lester444 Does this work for resuming after pausing for toolchange?

Lester444, 2018-07-18:
@Harvie I have not tried it. Easier for me to have separate g-code for each tool.

Harvie, 2018-10-29:
Program can now be started by cycle start HW button

closed 2018-10-29 by Harvie

Harvie, 2018-10-30:
yes. if you hold activate the start pin on grbl for fraction of second it will act the same as if you press the start button in bCNC. you just need reasonably recent version of grbl and machine state must be idle.

Harvie, 2018-10-30:
Nice to hear that. I was so busy with developing this feature, so i haven't find the time to install such button on my machine yet LOL. I do more coding than milling lately :-D

einencool, 2018-11-01:
Hi Harvie,
Today I've tested the hardware button.
What should I say.
Yeah, now it works really fantastic.
I can start my jobs just by pressing the button and now I don't have to go to the laptop and start and Pause it every time....

Really great work :-)
Thank you for your continuous support

einencool, 2018-11-02:
Hi @Harvie ,

I've got one question, but I didn't know, if it is a new "issue" or not.
But with a Version from August I was working on till now, that issue wasn't there.

The first time I start a program everything is fine, but the next parts, the status bar won't work any more.

This is the status bar when the program is running the first time
[image]

after the first job, it shows me how long the program was running etc..
[image]

but when I start the next program, the status bar doesn't show anything
[image]
nothing about the buffer, or how long the program will run etc...

can you help me in this case?

This is a message when I close bCNC
[image]

Thank you in advance
Greets Chris

Lester444, 2018-11-05:
@Harvie , dunno why but I can no longer see "Config", "Controller", "Shortcut", etc.  These used to be under the "Tools" tab. 

Is it because I am using "Double Size Icon"?  (I need big icons for my touchscreen monitor...... & my stubby fingers)

I had to go back to a previous version of bCNC (I can't see the version number on my screen) which I have been using June 2018. I hope you can fix this since the hardware "Run" button is working great for me.

Harvie, 2018-11-05:
@einencool @Lester444 can you please delete this and create new issues for it? there's  obviously something wrong, but it's unrelated to this issue.