vlachoudis/bCNC #899: Reset input control

neomadic, 2018-07-02:
I've been going through the learning curve on setting up a DIY plasma table and I'm using the protoneer v2.58 board.  I'm having trouble figuring out which input on the protoneer board to use for my e-stop and collision sensors.  ABRT stops motion and torch, resets everything, loses position, defaults back to mm in jog control(so annoying!) and usually locks up bCNC so I have to restart the whole rPI to get motor response again.   RST stops motion and torch and but loses the WCS zero by at least a few mm due to the gantry coasting a bit after the stop trigger.  Since the anti collision head uses light springs to locate the torch, triggering it usually doesn't result in position loss as the torch springs back into position after it bounces off the dross, tip up, or failed/alarmed probe cycle.

Is there a way to hook a switch to the HLD input on the grbl shield and simultaneously initiate a pause in bCNC and stop the torch/spindle?  

I can throw in an extra relay to mechanically disconnect the torch on a pause but a software option would be nice here too.

Harvie, 2018-07-08:
Probably related to #773

Harvie, 2018-10-30:
This is GRBL related issue. But i think you can use "feed hold" pin.

closed 2018-10-30 by Harvie