vlachoudis/bCNC #901: Slicing STL

Harvie, 2018-07-02:
There's STL slicer writter in python that would probably be good enough for integration into bCNC:

https://github.com/matthewelse/pySlice

All we need is to extract 2D outlines (layer cross cuts) from STL, so we can then use bCNC to create operations on such outlines.

Harvie, 2018-07-09:
There's very impressive python library called PyMesh:

https://pymesh.readthedocs.io/en/latest/index.html
https://github.com/qnzhou/PyMesh

It can Read/write 2D and 3D mesh in .obj, .off, .ply, .stl, .mesh (MEDIT), .msh (Gmsh) and .node/.face/.ele (Tetgen) formats.

https://pymesh.readthedocs.io/en/latest/basic.html#loading-mesh

It has simple API, that returns object containing paths for individual slices:

https://pymesh.readthedocs.io/en/latest/api_geometry_processing.html#slicing-mesh

I guess we can have basic 2D slicing in like 10 lines of code...

Harvie, 2018-07-09:
Another possibility: https://github.com/julienr/meshcut

Harvie, 2018-07-10:
I succesfully sliced mesh to g-code using meshcut library!

[image]

PoC code:

```python
"""
Basic usage example on a ply mesh. Note that this require a closed, manifold
input mesh.
"""
##
import os
import numpy as np
import itertools
import utils
import ply

import meshcut

if __name__ == '__main__':
    example_fname = "/home/harvie/Temp/meshcut/examples/data/mesh.ply"
    with open(example_fname) as f:
        verts, faces, _ = ply.load_ply(f)

    plane_orig = (0.02, 0, 0) #vyska
    plane_norm = (1, 0, 0)

    contours = meshcut.cross_section(verts, faces, plane_orig, plane_norm)

    for contour in contours:
        print("(---------------)")
        #print(contour)
	first = contour[0]
	print("g0 x%f y%f"%(first[1],first[2]))
	for segment in contour:
		print("g1 x%f y%f"%(segment[1],segment[2]))
	print("g1 x%f y%f"%(first[1],first[2]))
```

Harvie, 2018-07-10:
I've got first version of plugin!
[image]

Harvie, 2018-07-11:
Currently i am trying to solve problem with holes not being sliced at all: https://github.com/julienr/meshcut/issues/3

Harvie, 2018-07-11:
Now we have two modes.
Non flat:

[image]

Flat:

[image]

julienr, 2018-07-14:
This looks pretty cool and I'm happy if my small meshcut library helps :)

Harvie, 2018-08-18:
In #937 

Added basic mesh manipulation to slicer plugin:

[image]

Added 3D slicing (useful for future 3D cam work):

[image]

[image]

Harvie, 2018-08-18:
@julienr in fact i've just found bug when slicing your sphere.stl in Y layers 0, -2, -4 and possibly more:

[image]

it's for the following settings:

```
plane_norm = (0, 1, 0)
#and three of these:
plane_orig = (0, 0, 0)
plane_orig = (0, -2, 0)
plane_orig = (0, -4, 0)
```

Directly in bCNC you can launch it with these settings:

[image]

Do you think you can check what's going on?

julienr, 2018-08-29:
I tried to reproduce in bCNC, here is what I do :

- open the Slice Mesh plugin
- set the parameters as on your screenshot
- Click on 'Slice Mesh' to run the plugin

And then it seems to run for a while, but afterwards I don't see any result in the 3D view. Do I have to click somewhere else ?

I tried to reproduce using the `1_stl_sphere_cut.py` example of meshcut, but it seems to work there :

[image]

(This is with plane_norm = (0, 1, 0))

I have to use plane_orig with y between 0 and 1 though because otherwise, the plane doesn't intersect the sphere. Maybe in bCNC the sphere has a different origin / scale ?

Harvie, 2018-08-29:
@julienr i am intersecting the sphere from side. the stl is actually hemishere. as you can see in meshlab:

[image]

problem happens if i do several slices from the side:

[image]

julienr, 2018-09-06:
The gap in the screenshot is just that the last point is not connected to the first when drawing, so I don't think that is the issue.

I tried to reproduce the issue in bCNC, but I am a bit lost (I haven't used bCNC before). Could you give me exact steps to reproduce the results you get ? When I run the slice mesh plugin, all I get is a sphere but I don't see the slices like you do.

Also, how can I show the yellow list of slices like on your screenshot ?

Another option would be if you can reproduce this with a script using the `sphere.stl` that is included in meshcut, that might help with debugging.

Harvie, 2018-09-06:
@julienr click on the editor tab:
[image]

Harvie, 2018-09-06:
I've tried with sphere.stl from meshcut repository and problem is very similar. Here are my settings:

[image]

eg. layer y0.75 is very bad:

[image]

Harvie, 2018-09-07:
I've just found hacky way to produce 3D g-code from 3D STL. It's FAR FAR from precise, but it was so easy to do, so i just had to. Now i mark all Z slices as islands. That means when i do flat pocket or "flatten" and cut it, the islands cause the 3D model to get imprinted to that flatten code:

[image]

The result looks terrible, but i've managed to add this feature in like 10 minutes :-)

[image]

julienr, 2018-09-08:
@Harvie Thanks for the explanations, managed to reproduce in my meshcut example ! I opened https://github.com/julienr/meshcut/issues/7 to not pollute this issue with my debugging progress. I'll notify you here when I manage to fix it.

Harvie, 2018-09-08:
@julienr Thank you very much!
BTW i am working on lowering number of dependencies of meshcut. Currently i managed to get rid of some of them. Eg.: here i managed to calculate spatial distance using numpy vectorization rather than scipy:

https://github.com/Harvie/bCNC/blob/master/plugins/slicemesh.py
(I would call this plugin "meshcut", but bCNC imports the plugins same way as libs, so there was conflict, which leaded to meshcut library imported rather than plugin file itself, so i had to rename it to slicemesh in order to make it possible to import both library and plugin)

And managed to remove need for "enum" library in stl.py:

https://github.com/Harvie/bCNC/blob/master/lib/stl/stl.py

My longterm plans are to minimize the infrastructure needed to parse PLY and STL files and add support fro binary PLY files (currently only ASCII PLY are supported).

julienr, 2018-09-17:
@Harvie I indeed found a bug in the slicing code and pushed some fixes to master. Should work better now.

Harvie, 2018-10-30:
STL slicing is implemented. If there's ever will to implement 3D CAM on it, we can open separate issue.

closed 2018-10-30 by Harvie