vlachoudis/bCNC #902: SVG import

Harvie, 2018-07-02:
This looks like it would be easy to integrate to bCNC:

https://github.com/vishpat/svg2gcode

SteveMoto, 2018-07-03:
pay attention:
https://www.gnu.org/licenses/gpl-faq.html
[image]
Greetz
Steve

Harvie, 2018-07-03:
I guess we can ask original autor to relicense for us. Chances are he just grabed GPLv3 without any specific reason... :-)

Harvie, 2018-07-12:
Another interesting project https://github.com/mattdesl/svg-path-contours

Harvie, 2018-07-14:
Another one: https://github.com/SebKuzminsky/svg2gcode/

Harvie, 2018-08-17:
[image]

SVG export:

[image]

Harvie, 2018-08-18:
These guys also have some python code for SVG path linearization:
https://pypi.org/project/SVGCompress/#files

Harvie, 2018-08-18:
I've removed everything unneeded from the library:
svg_gcode.py:

```python
import numpy
import svg.path
import lxml.etree as ET
import os

'''
Compress svg figure through a variety of methods
Written July 22, 2014
Author: Gen Del Raye
'''

class SVGcode:
    '''
    Read in the paths in a svg file and compress selected polygons, then output the compressed
    svg to file. The general workflow should follow:
        comp_instance = compress(filename)                # Read svg
        comp_instance.find_paths()                        # Extract path coordinates
        comp_instance.linearize_paths(curve_fidelity)     # Convert paths into polygons
        comp_instance.compress_by_x(*args, **kwargs)      # Compress polygons ( where 'x' = 'merging',
                                                          # 'deletion', or 'simplification').
        comp_instance.write(outputfile = outputfile)      # Output compressed file
        comp_instance.report()                            # Report on compression stats
        
    Most of this functionality can also be accessed through a convenience function 'compress_by_method()'
    See usage examples in test()
    '''
    def __init__(self, svg_file):
        '''
        Read svg from path 'svg_file' (e.g. 'test/test_vector.svg')
        '''
        #self.filename, self.extension = svg_file.split('.') # Get filename and extension
        #assert self.extension == 'svg', 'File must be an svg'
        try:
            self.figure_data = ET.parse(svg_file) # Parse svg data as xml
        except ET.XMLSyntaxError, e:
            '''
            Large svgs may trigger lxml's 'excessive depth in document' exception
            '''
            warnings.warn('lxml error: %s - Trying huge xml parser' %(e.message))
            huge_parser = ET.XMLParser(huge_tree = True)
            self.figure_data = ET.parse(svg_file, parser = huge_parser)
        self.root = self.figure_data.getroot() # Root object in svg
    
    def _get_tag_prefix(self):
        '''
        Return the string that prefixes all tags, 
        e.g. if the root tag is '{http://www.w3.org/2000/svg}svg', all other tags 
        in the xml are probably prefixed by '{http://www.w3.org/2000/svg}'
        '''
        # TODO: Fix this hack
        prefix_endindex = self.root.tag.rindex('}')
        return self.root.tag[:(prefix_endindex + 1)]

    def find_paths(self):
        '''
        Find and parse nodes in the xml that correspond to paths in the svg
        '''
        tag_prefix = self._get_tag_prefix()
        #tag_prefix = '{http://www.w3.org/2000/svg}'
        #tag_prefix = '{[^}]*}'
        self.path_nodes = self.root.findall('.//%spath' %(tag_prefix))
        self.path_parents = self.root.findall('.//%spath/..' %(tag_prefix))
        self.paths = list()
        self.paths = [svg.path.parse_path(p.attrib.get('d', 'M 0,0 z')) for p in self.path_nodes]
        
    def linearize_paths(self, curve_fidelity = 10):
[... 60 lines cut ...]
            linearized = segment_linearize(segment, n_interpolate)
            complex_coords.extend(linearized[:-1])
        # Append last point of final segment to close the polygon
        complex_coords.append(linearized[-1])
        return [self.complex2coord(complexnum) for complexnum in complex_coords] # Output list of (x, y) tuples
```

test.py

```
from svg_gcode import *
comp_instance = SVGcode("test/test_vector.svg")
comp_instance.find_paths()                        # Extract path coordinates
comp_instance.linearize_paths(curve_fidelity = 10)     # Convert paths into polygons
#comp_instance.write(outputfile = "hrv.svg")
#print(comp_instance.linear_coords[0])
for path in comp_instance.linear_coords:
	p = path[0]
	print("g0 x%s y%s"%(str(p[0]),str(p[1])))
	for p in path:
		print("g1 x%s y%s"%(str(p[0]),str(p[1])))
	print "( ---------- cut-here ---------- )"
```

still needs some cleanup, but 130 LoC is nice

Harvie, 2018-08-21:
I was unable to contact the author of SVGCompress, so i ended up writing my own SVG2Gcode converter, which is even simpler!

Harvie, 2018-10-30:
Implemented

closed 2018-10-30 by Harvie