vlachoudis/bCNC #903: Rounding problems

Harvie, 2018-07-03:
When i generate inside profile of rectangle with overcuts and then try to do outside profile of this profile it leaves out segments around overcuts due to fact these are not rounded properly. When i manualy round the line it gets fixed. Also when i use Editor -> Edit -> Round tool.

[image]

Harvie, 2018-07-08:
This fix caused different problem:

[image]

Harvie, 2018-07-08:
To sum the problem:
  - when i don't round the blocks, some segments are missing from profile
  - when i round the blocks some arcs are repeated twice in profile

@vlachoudis any ideas pls?

Harvie, 2018-07-08:
it was working at 6b468a573e18f47d473fb8645a6366ef5456defc
and broken at 868b04dad169cabf76740cf5a0c48ffe170851dd
so the bug must have been introduced somewhere between...

Harvie, 2018-07-08:
Can you please reopen this? It was not really fixed by my commit. i reverted that later. We have to sort this out...

vlachoudis, 2018-07-08:
Sorry it was by closed by mistake. I will try to reproduce and see what is going on.

Harvie, 2018-07-08:
Thank you. Here is the test file for the second case:

```
(Block-name: Header)
(Block-expand: 1)
(Block-enable: 1)
M3 S12000
G4 P3
(Block-name: 0)
(Block-expand: 1)
(Block-enable: 1)
G0 X5.257 Y8.5573 Z0
(entered)
G1 Z0 f200
G1 X7.2021 Y-13.2064
G1 X-8.4555 Y0.1244 Z0
G2 X-4.7178 Y9.2661 I3.3693 J3.9574 Z0
G1 X-2.7228 Y9.1244 Z0
G3 X1.2671 Y8.8409 I1.995 J-0.1418 Z0
G1 X5.257 Y8.5573 Z0
(exiting)
G0 Z10
(Block-name: Footer)
(Block-expand: 0)
(Block-enable: 1)
M5
G0 Z20
```

[image]

walterwissmann, 2018-08-13:
I have had also this effect. But i am going back to LibreCAD and make the DXF Line absolut close...
I think bCNC should be a little bit tolerantly with minimal open lines.

Harvie, 2018-08-14:
@walterwissmann AFAIK it happend to me on lines that came from LibreCAD without any gap. We might try to tweak the bCNC's method which tells if two points are equal within tolerance. But i am not sure if that's the problem.

vlachoudis, 2018-08-14:
bCNC is quit strict on tolerances. It is better to leave it like this, because no matter how much you relax it, there will be always a case that will need more.

walterwissmann, 2018-08-14:
The lines were not together in a really small area.
Often the problem occurs when rounding corners...
But I also think the tolerance function of bCNC is ok as it is

Harvie, 2018-08-14:
Yeah, but temporarily lowering the precision for debuging might get me some insight on what goes wrong. There's something wrong and it blocks #220. Especialy when bCNC fails to process g-code that was generated by itself. I think bCNC should round the paths to the precision which can be handled by it. But this is really strange. When it's too precise it's broken, but when it's not enough precise, it's still broken, but differently. We need to figure out what happens.

walterwissmann, 2018-08-14:
@Harvie, Ok...
I have tell this to my partner... And we found:
The incoming and the outgoing from the ground-database are the main problem...
around there shouldt we found the round problem... Why we think so?
Now:
it must be in bcnc a database independently from others...
From there bcnc can generade the output to the screen, the input from a DXF the output to the gcode and so on...

Around this easy database are the round problem...

Harvie, 2018-08-28:
Note to self: Maybe it's related to following code in CNC.py profile():

```
                                # Remove tiny segments
                                path.removeZeroLength(abs(offset)/100.)
                                # Convert very small arcs to lines
                                path.convert2Lines(abs(offset)/10.)
```

commit 905130a0 (2018-07-08): Disabled the rounding again. Have to fix #903 in different manner.