vlachoudis/bCNC #912: Reimplement tabs using islands

Harvie, 2018-07-08:
At first the islands were implemented using heavily modified code for tabs. Now it's time for tabs to be implemented using islands as idea of islands is more generic than idea of tabs. Also currently tabs can't be moved or edited. I suggest following solution:

Implement some primitive way of block grouping, so that whenever you select one block, it will automaticaly select other blocks that are linked to it.

That way you can generate tabs as g-code blocks that will get attached to one or more toolpaths and therefore they will automaticaly get selected, moved and cut with that toolpath (unless user disables islands in cut operation).

This will enable us to:
  - move tabs, modify them much more than was originally possible
  - remove special rendering and handling of tabs (we can reuse block manipulation same way as islands do)
  - expose grouping to users. eg. for cases when you need to repeatedly do operations on multiple groups of blocks and don't want to select them again and again

What do you think? Any ideas?

Harvie, 2018-07-08:
Partialy done
[image]

Harvie, 2018-07-08:
This was easy fix. Everything works, and it's possible to move and edit tabs!
I realized that there's no really need for grouping as i can put all the tabs into single block, so it's easy to select all related tabs at once.

Only thing that is left to do is to remove LOTS of tab related code that is now deprecated.

vlachoudis, 2018-07-09:
@Harvie the tabs implementation was offering the move capability. When you expand a block of code and select move, you could position anywhere the tabs. Also it was offering editing capabilities from the text editor, since you could see the tabs as plain text. I like the idea of islands, but I am not happy having a hex-ascii string not human readable emdeded in a plain ascii gcode file as it is your implementation now.

Harvie, 2018-07-09:
@vlachoudis have you seen lastest pull request??

Harvie, 2018-07-09:
It was possible to move whole block of tabs at once. Now i made block split/join infrastructure which allows individual tabs to be easily moved independently:

[image]

closed 2018-10-30 by Harvie