vlachoudis/bCNC #914: Problem in trochoidal splicing straight

MARIOBASZ, 2018-07-09:
In splicing straight leaves the route when direction changed
[image]

Harvie, 2018-07-09:
This was already fixed in PR #911 .

MARIOBASZ, 2018-07-09:
Ok,

Harvie, 2018-07-09:
It is the place, but i actively work on trochoidal and chances are that by the time you file issue, there will already be new unmerged version in my fork. So if you want to try the latest version of trochoidal, you can watch my fork directly, so you don't have to wait until merge...

Harvie, 2018-07-16:
I added support for following ARC paths by trochoids. You can discus trochoids in #815 and close this issue...

closed 2018-10-30 by Harvie