vlachoudis/bCNC #916: Stops sending gcode - Spindle stays on (CNC Router running Smoothieware CNC build)

ccomito1223, 2018-07-11:
This is brand new to me. All was working until the 3rd time I executed this job. My setup is smoothieboard with latest cnc config setup for grbl mode. I've tried multiple computers and multiple gcode files. Same results but it worked fine an hour before the initial crash. I also think this has been addressed but I'm to new to be sure. Please point me in the right direction.

bCNC V0.9.14

Exception in thread Thread-2:
Traceback (most recent call last):
    File "C:\Python27\lib\threading.py", line 754, in run
         self.__target(*self.__args, **self.__kwargs)
    File "C:\bCNC-master\Sender.py", line 853, in serialIO
         self.serial.write(b"?")
    File "C:\Python27\lib\site-packages\pyserial-3.4-py2.7.egg\serial\serialwin32.py", line 323, in write
          raise writeTimeoutError
SerialTimeoutException: Write timeout

Giochia, 2018-07-11:
I do not know if this can be similar, but I had a random problem too and it drove me mad until I found that it was the spindle motor that was introducing a lot of "electrical noise", thus interfering with serial communication and with other electronics. It was sufficient to add 3 capacitors to the motor, to kill most noise and all worked fine ever since.
So I suggest that it may be an idea to check for electric noise first. If you have tools to check this, like an oscilloscope, this may be easy, if not you will have to find some workaround. One easy way is to lift the Z axis, so as to ensure it will not touch, and run the job without the workpiece and with the spindle off. If it doesn't fail, it may be it. Do the same with the spindle on and see if it fails, if yes, then there you have it!

MARIOBASZ, 2018-07-11:
Can you draw the circuit with the values of the capacitors you have used? (or is it more complicated to determine the values?) I have this problem of electrical noise when machining aluminum with a spindle of 500w

Giochia, 2018-07-11:
Hi, sorry, I have no sufficient knowledge to give you such information, but you can find dozens of ideas online. Such as this: https://www.pololu.com/docs/0J15/9   or this: http://www.recentscientific.com/sites/default/files/2485.pdf   or this: https://www.superdroidrobots.com/shop/custom.aspx/motor-wiring-support/19/
I also had to add ferrite rings on the wiring from power supply.

ccomito1223, 2018-07-12:
Hello @Giochia I will give this a try but I don't think I have all that much noise. I could but I'm using very high quality shielded cables on most everything. Not to mention that this issue appeared out of nowhere and everything was fine for days prior. I'm going to pull apart my control box tonight and see if it's perhaps a loose ground or something. Thanks for the input though..

Lester444, 2018-07-18:
I had this same problem when I switched over to an original Arduino UNO (16U2). The feed stops somewhere in the g-code and the spindle keeps running. It's a nuisance for code that takes more than 10 minutes to run because I now have to re-run the g-code. Usually happens the first time I run g-code from a fresh re-start of bCNC,, but doesn't re-occur on succeeding runs.

The problem disappeared after I went back to an Arduino clone (CH340G).

Running bCNC on a Raspberry Pi 3.

I am waiting for the new release of GRBL (on ARM I think) so hopefully this problem will  go away by then.

ccomito1223, 2018-07-18:
I might have resolved the problem. In my control box I have a usb feedthrough coming from the controller to the outside. I disconnected that and ran the job again. I didn't have a single drop out in 4 runs. I'll test more tomorrow but I'm hoping that was the problem.

ccomito1223, 2018-09-29:
Just to update the issue, the problem still persists. I thought it was resolved but it still comes back. It's very random when it happens. Some days there are no issues at all and other days (like today) I can't complete a single task. It's very frustrating. I'm going to try something else (don't know what exactly but something)... Suggestions are welcome!!!

Giochia, 2018-09-30:
Have you made sure the motor does not introduce electrical noise ?
Maybe you could try this: https://www.pololu.com/docs/0J15/9