vlachoudis/bCNC #918: Help Adding Block Name to self.msg

jpagliaccio, 2018-07-12:
For my use of bCNC, it would be great to have the current Block Name appended to the end of self.msg in lib/tkExtra.py. I haven't been able to get that working. Could someone supply a little direction or the code that would do that?

If I had Block Name I could pause and then restart  work  easily, or restart when things hang up. The current line is difficult to identify in the same way. 

Thanks in advance.

Harvie, 2018-07-15:
There is one caveat. GRBL does not execute received g-code immediately. AFAIK it keeps few lines in cache, so if you know that bCNC sent some line to GRBL it does not really mean that it was cut.

jpagliaccio, 2018-07-22:
Thanks. I understand that. But I still would like to see which the block name when there is an issue or need to restart. 

Any Python help adding that to self.msg would be appreciated.

Harvie, 2018-08-02:
This should be pretty easy to implement. I might add this feature, but currently i am waiting for some of my previous modifications to be merged to upstream and i don't want to add new stuff untill that is figured out.

jpagliaccio, 2018-08-04:
Tomas:

Thanks. Let me know if I can help test or something.

Joe