vlachoudis/bCNC #920: Need new interface

avilab84, 2018-07-15:
Hello @allDevs! This small issue is complementary to another. I show you what I would like and explain why!
I would like an interface like this:
1 With no complicated false 3D icons, only flat icons or semi flat.
2 no more than two colors to be clearly understood at the first glance.
Two colors create an coherent and simple graphical language so that the brain can quickly understand the priority and the important elements of an icon.
3 Dark theme, because it has been proven by study that the brain can read faster and understand better when withe is written on black, and this is also better to see in the distance.
4 Not 1995 looking tabs but 2018 ones please, this is a serious problem of this software, A lot of peoples came to use the machine that I sell and don't buy because they are not COMFORTABLE with the software, the word comfortable is important here because I know the software works but it is not comfortable.
5 Please use wide spread icons if the exist and self-explanatory ones so that a 10 year old can understand them, the spindle icon is a good example. Good joke, but I would prefer a mill icon instead! You know I helped you to change the USB icon that was a VGA cable, maybe I will find the time to do the rest, but please IF YOU ARE A GRAPHIST DESIGNER, HELP US ;)
[image]
[image]

Harvie, 2018-07-16:
I am well aware of this situation. First thing that has to be done before we can get proper UI is rewriting CNCCanvas code to OpenGL. #591

Harvie, 2018-08-21:
@wmarcondesbr :+1: pull requests are welcome