vlachoudis/bCNC #921: Approximate large number of small lines using arcs

Harvie, 2018-07-16:
This is more of a long term goal. I'd like to have plugin which would be able to take toolpath which has arcs and circles stored as lots of small lines and replace these with arcs to lower g-code size.

I've heard that some STL slicers used in 3d printing already can do this (they use this to optimize g-code, because STL slices are always lines sliced from triangles).

Harvie, 2018-08-31:
These guys can do it already (under GPLv3):
http://mythagel.github.io/nc_tools/nc_arcfit/
https://github.com/mythagel/nc_tools/tree/master/src/nc_arcfit

Harvie, 2018-09-06:
Also this maybe:
https://github.com/blinkenlight/LineBender

Harvie, 2018-09-07:
Another GPLv2 python code which does this:
http://wiki.linuxcnc.org/cgi-bin/wiki.pl?LinesToArcs

Harvie, 2018-09-29:
I've got basic functionality!
[image]

sfinexer, 2018-09-29:
@Harvie 
I took the code, and that's what happened.

> Y0X0Z0
> G1F1000
> X-20F1000
> Z-10F1000
> Z0F1000
> Y20F1000
> Z-2F1000
> X-10F1000
> Y10F1000
> X-20F1000
> Y20F1000
> Z0F1000
> Y0X0F1000

[image]

Harvie, 2018-09-29:
@don't worry about this. this is caused by bug that's not related to arc fiting. it affects more features when there is path consisting of several loops. i'll have to take a look at it.

Harvie, 2018-09-29:
@sfinexer Update: your g-code has Z coordinates in it. It's not gonna work for now as lib/bpath can't handle Z. sorry. Also whole algorithm would be bit more complex to handle Z.

sfinexer, 2018-09-29:
@Harvie, So it is necessary to specify in the options the exception of Z coordinates.

Harvie, 2018-09-29:
How do you clean spindle using g-code? :) And why do you need to arcfit spindle cleaning g-code which has no arcs in it?

sfinexer, 2018-09-29:
@Harvie  Very simple.
positions the Z axis is the work area (negative X coordinates) and spend on tip grid.
But I need arcs to smooth the code I got sketch.py

Harvie, 2018-09-29:
Anyway. I am not really willing to develop any 3D functionality in bCNC until #591 is fixed. I see no point in having advanced 3D features in UI which can't properly render 3D toolpaths, it can be really annoying...

It should take g-code from sketch.py... I guess there's some problem related to multiple paths in single block. i'll try to take a look at it. but not sure when. problem seems to lay in Block.toPath()

sfinexer, 2018-09-29:
Made 2D.
I want to smooth the corners.
[image]

Harvie, 2018-09-29:
I've fixed the problem. But i think  that you've misunderstood what this plugin does!
It's arc fiting plugin. Not corner smoothing. If you need to smooth the corners you have to do it in the CAD.

Harvie, 2018-10-29:
http://www.maths-in-industry.org/miis/682/1/ESGI_104_P7.pdf

Harvie, 2018-10-30:
This is now in master, but it might still need some tuning.

closed 2018-10-30 by Harvie