vlachoudis/bCNC #922: BCNC not loading all tool paths

mogdustin, 2018-07-17:
The issue I am having is bcnc is not loading all parts of my Gcode cuts. It looks like it is only loanding a 1/4 of my toolpath. When I load the Outline cut of the sign the whole thing loads. When I load the Gcode part attached is what I get. it looks like it is being cut off at the margin. I made the tool path in aspire and am not sure what is happening or how to fix it. 

[image]

toktuff, 2018-07-18:
It might be that the drawing of your tool path timed out. It will not display the entire tool path in that case. See if increasing the timeout in the dropdown at the top-left of the main drawing area helps.

Harvie, 2018-07-19:
This might get better once #591 fixed :-)

closed 2018-10-30 by Harvie