vlachoudis/bCNC #923: Invalid code when MOVE/ORIGIN used with inline plane commands (G17,G18,G19)

Bologrew, 2018-07-29:
I used the Fusion 360 GRBL post-processor to generate some G code and loaded it into bCNC.

This is an excerpt from the code:
%
X-32.445 Y-241.362
G1 Z-2.5 F100
G19 G2 Y-241.862 Z-3 J-0.5
G1 Y-242.362 F1000
G17 G2 X-32.945 Y-242.862 I-0.5

G1 X-47.565 F500
G19 G3 Y-241.362 Z-2.5 K0.5
G0 Z5

M30
%

I then used the "ORIGIN" command to move all the code ("MOVE" does the same).

This is the result:
(Block-name: Header)
(Block-expand: 1)
(Block-enable: 1)
%
X-50.1433 Y-243.57429
G1 Z-2.5 F100
G2 G2 Y-244.07429 Z-3 J-0.5
G1 Y-244.57429 F1000
G2 G2 X-50.6433 Y-245.07429 I-0.5

G1 X-65.2633 F500
G3 G3 Y-243.57429 Z-2.5 K0.5
G0 Z5
(Block-name: block)
(Block-expand: 1)
(Block-enable: 1)

M30
%

This code is no longer valid because it contains two G2 or G3 commands on a single line.
The plane selection commands G17, G18 and G19 have been lost.
I checked and you are allowed to have a plane selection G code and a motion G code on the same line as they come from different groups of commands.
These sorts of movements are created when using lead-ins and lead-outs in Fusion 360.

I'm no Python expert but I have tracked the error down to the "process" function in "CNC.py".  The code only expects to find one "G" command on a line so the plane selection one gets overwritten by the later motion one.

May I make one further suggestion - when importing code generated elsewhere, it would make it much easier if the automatically generated blocks were given an incrementing number.

Despite this bug, bCNC is a very good piece of software and I have been running it on a Raspberry Pi 3 with a Protoneer CNC controller piggy-backed on it to control both a CNC router and a desktop milling machine.

Bologrew, 2018-07-30:
Oh that's frustrating!
The first thing I did when I found a problem was to check the "Updates" button.  It said I was running version 0.9.11 so I clicked the "Check now" button and it said the latest GitHub version was also 0.9.11.
However, I have now downloaded the latest zip version from the site and it says it is 0.9.14 and the code that was giving me the problem appears to have been rewritten.  So that's good news.

However, when I click the "Updates" button and "Check now" on this latest version, it still tells me that the latest GitHub version is 0.9.11.  So perhaps the bug is actually in the updates checking code!

So it's time to test out this latest version and it looks like there's a few new bells and whistles to play with.

Harvie, 2018-08-03:
Dunno. Maybe 0.9.14 is development version, but not official stable release yet, but 0.9.11 is stable release?