vlachoudis/bCNC #926: Semi threads in y plane

MARIOBASZ, 2018-08-05:
I'm thinking of making a complement that allows to make semi threads in xy plane

[image]

Harvie, 2018-08-07:
This will be very hard to do effectively, because of endmill geometry.

MARIOBASZ, 2018-08-07:
I know that it is not the ideal way, there will be a concave seam, which can be improved with a round tip milling cutter

Harvie, 2018-08-07:
You can mill perfectly round **valley** all around using ball nose endmill, but how do you make round **hill**?

MARIOBASZ, 2018-08-07:
As I have reasoned, it is possible quite simpe! I am rewriting my trochoid 3D and in parallel trying the same on the base of yours. Can you make a version where you take the values ​​in X and Z instead of just XY? (I have made a modification, but I find this problem: In circular paths, goes by the rope in place of the arch, surely choose the best route as the straight) It will probably be easier and faster to you. From there we are simply one step away.
It is interesting to be able to lean on someone and add skills to work together.
I do not know if you have seen the trochoid Path, it would be necessary to optimize the warning form to start the path with adaptive radius trochoids

MARIOBASZ, 2018-08-07:
The hill will certainly not be round

MARIOBASZ, 2018-08-07:
Moving forward now with 3D trochoid !!

MARIOBASZ, 2018-08-11:
correction: with one more pass with bit more girl I get rounded hill

MARIOBASZ, 2018-10-14:
Advances
[image]