vlachoudis/bCNC #928: Setting the homing zero

DLMiller5, 2018-08-08:
As my CNC machine homes it will hit the switch back and then travel to Mpos. The Mpos for "X" and "Y" are stopping 1" in to the work area. What that means is that I am losing 1in "X" and "Y" in work space. How can I adjust the "X" and "Y" to stop where I can maximize my work area?

walterwissmann, 2018-08-11:
I think that is a Configuration-Problem with grbl you have. 
See $24 to $27 and 130 to $132 on:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

DLMiller5, 2018-08-14:
I played around with $24 to $27 and $130 to $132 with no change. The Home
Zero "X" and "Y" always homes @ -1 or 1 inch into the Work space. The "Z"
is just fine I can set the Work Zero anywhere within the work space and
then zero the 3 axes. This functionality doesn't matter if I use bCNC or
GCS. I know there should be a GRBL setting to limit the bounce back once
the Home switch has been closed.

Questions

   1. When the Scan button is pushed is the spindle suppose to follow the
   values set in #130 and #131, or the outer board for the work area?
   2. When I supply power to the RPi CNC controller isn't the steppers
   suppose to lock?

*Dennis Miller*

walterwissmann, 2018-08-15:
1.:
I think you pushed the Home bottom...
grbl search then the home position and set the wpos.
Now the workspace is little bit smaler: workspace in $130 to $132 minus $27. 
See also:
https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle
And there: Home travel

2.:
I know nothing about this board... It is a question to the dealer ;)

DLMiller5, 2018-08-15:
Thank your for your reply

Dennis Miller

closed 2018-10-30 by Harvie