vlachoudis/bCNC #929: 2d digitizer, patern digitizer, jog digitizer.

FatNarval, 2018-08-09:
Hello,  great work on bcnc!
I personaly like having camera probing, it is great feature!

It is great for simple shapes, I can manualy copy (with mouse)  x,y coordinates and create gcode or plt file which could be exported and edited as vectors in other cad or cam programs.
But for  it is very tedious and error prone process due to manual copying of coordinates.

 I wish that  bcnc could be easier to use as 2d pattern digitizer,  you jog to a point pres a button/key and coordinates are stored in a single text file.  Repeat. That simple. Is it actually possible to implement this feature?
Maybe it could have options to save file as a Gcode or plt file or dxf or xls?

If this option catches on,  conversational options  like: arcs, circles,  g0/g1, feeds, Z position...etc could be added later.

Thanks

MARIOBASZ, 2018-08-10:
uau! It would be good! I do not know how hard it is, I do not know if I am reasoning perfectly, but another application could be combined with visual recognition, it would be another way to find limits of a piece, hole center, or alignment for two-sided pcb.

MARIOBASZ, 2018-08-10:
Or 2D 3D scanner

FatNarval, 2018-08-10:
@MARIOBASZ   Nothing that fancy :)  I have in mind very simple solution using existing hardware (and software).
 For jogg digitizing camera is not even neccessary but it helps, you can do all digitizing fron your desktop. 
I do not know how complicated it is as I am not a programmer myself
I can not find good ilustrations on the net but here are couple of them, they are using encoders and manualy move the pointer. Idea is the same just use jog moves and camera .
https://youtu.be/lPonmGaA7Mw
https://vimeo.com/17095655
https://www.youtube.com/watch?v=hVMaa1NueWI

FatNarval, 2018-08-11:
This is best ilustration for what I had in my mind, it is aplugin with simmilar features for mach3. I can not find any yt video to ilustrate how it works.
https://www.machsupport.com/forum/index.php?topic=28901.0

FatNarval, 2018-08-11:
@MARIOBASZ you orobably had something like this on your mind?
https://youtu.be/vTrGxb7s1oc

MARIOBASZ, 2018-08-12:
It would not be bad if someone wants to work on it! Perhaps it is very ambitious, with an emitting diode and a phototransistor as a line follower or complement with artificial vision. By the way, I do not know if a diode and phototransistor could be used to generate autolevel points

Harvie, 2018-09-03:
I've added basic 2D digitizing feature to Probe page:
[image]
Just jog and click to record the point!

FatNarval, 2018-09-05:
Wow! Great!
That was just I had in mind!

Testing soon!
Thanks!

swarfer, 2018-09-20:
please refer to 
https://github.com/vlachoudis/bCNC/issues/967
where I have provided code for using shortcut keys for the RAPID and FEED buttons

closed 2018-10-30 by Harvie

FatNarval, 2018-12-01:
Is it possible that control (jog) pad that is featured in control tab be also featured in probe tab?
This is very important for manual digitizing to be able to quicklyand change jogging step length .
Thanks!

Harvie, 2018-12-02:
@FatNarval this is already proposed in #1057