vlachoudis/bCNC #935: I face an issue and can't launch bCNC

tmvv2001, 2018-08-15:
Here is the error code.
Is any meet this issue before? 

Traceback (most recent call last):
  File "C:\BCNC\bCNC.py", line 2592, in <module>
    application = Application(tk)
  File "C:\BCNC\bCNC.py", line 113, in __init__
    self.ribbon = Ribbon.TabRibbonFrame(self)
  File "C:\BCNC\Ribbon.py", line 494, in __init__
    image=Utils.icons["new"],
KeyError: 'new'

alk1913, 2018-08-15:
hello,i have the same issue.this is the first time than i have this error.
i uninstall python (2.7.15),and reinstall,but the issue persist.
thanks for your help.

tmvv2001, 2018-08-16:
Hi...
I change to bCNC-0.9.10....then it work.

alk1913, 2018-08-16:
yes,i get it work,but the current code available on github doesn't work for me.
i can't open bCNC since that the icons change from gif to png.

gxurma, 2018-08-17:
Same issue for me.
Fresh clone.

C:\Users\gyurma\Documents\bCNC>bCNC.bat
Traceback (most recent call last):
  File "C:\Users\gyurma\Documents\bCNC\bCNC.py", line 2592, in <module>
    application = Application(tk)
  File "C:\Users\gyurma\Documents\bCNC\bCNC.py", line 113, in __init__
    self.ribbon = Ribbon.TabRibbonFrame(self)
  File "C:\Users\gyurma\Documents\bCNC\Ribbon.py", line 494, in __init__
    image=Utils.icons["new"],
KeyError: 'new'

alk1913, 2018-08-17:
i change the icons extension for gif and change the lines in pendant.py and  utils.py from png to gif and i add centerpoint.gif and flatpat.gif to the icons folder,with this changes i can launch bCNC.

jetaudio, 2018-08-19:
can you please tell me more specifically about how to fix this issue? i'm facing the same issue with newly downloaded clone

gxurma, 2018-08-20:
I changed back to a previous version, before the latest merge. That works.

alk1913, 2018-08-22:
[bCNC-master (22).zip](https://github.com/vlachoudis/bCNC/files/2312136/bCNC-master.22.zip)
try this

Harvie, 2018-09-10:
This is already fixed in my branch https://github.com/harvie/bCNC/
Hopefully the @vlachoudis will accept my PR #937 to get this fix upstream

Harvie, 2018-10-30:
fixed

closed 2018-10-30 by Harvie