vlachoudis/bCNC #936: Why is zstep disabled by default config?

Harvie, 2018-08-15:
Currently the default bCNC.ini looks like this:

```
[Control]
step   = 1
wcs    = 54
;zstep = 1
...
```
Is there some reason why zstep is disabled, which makes it inacessible from UI until user manualy changes the file. I don't see any downsides of enabling it by default. It adds one more UI element, but usualy you use * / + - buttons to set step size, which affects both step and zstep at once, so you don't have to care about zstep unless you manuly change it using dropbox.

walterwissmann, 2018-08-15:
I have an extra wireless keyboard block on my raspy3 so i can go around the machine to manualy position the gantry. 
On this keyboard i found the +;- and *;/ to increase and decrease the steps.

In this case it is better for beginners to have the zstep disabled...

Harvie, 2018-08-15:
@walterwissmann i plan to use exactly same setup using wireless keyboard. but as i said. zstep and step are always both affected by these keys, no matter if you have zstep disabled or enabled.

Guenni75, 2018-09-13:
How can i disable zstep?
I changed .bcnc to ;zstep, saved and opened bCNC.
The ZStep is still another value than Step and the listbox of zstep is still there.
After closing bCNC I opened the .bcnc again and zstep is gone but still in the bCNC UI.
I always have same values in both boxes but with this option, sometimes the zstep changes after opening and closing bcnc. Thats frustrating!

Harvie, 2018-09-13:
@Guenni75 I guess i'll later add some GUI to enabling/disabling zstep

Guenni75, 2018-09-17:
@Harvie I don't Need something in the GUI.
I tryed to disable it in the .bcnc, but that does not work.
The zstep value (in the .bcnc file) was deleted after i changed zstep to ;zstep but in bCNC it is all the same.
The Z-Listbox is still there and the ZStep changes sometimes. Most of the times after probing Z!

Guenni75, 2018-09-18:
I changed the ControlPage.py:

def divStep(self, event=None):
		if event is not None and not self.acceptKey(): return
		step, power = ControlFrame._stepPower(self.step.get())
		s = step/10.0
		if s<_LOWSTEP: s = _LOWSTEP
		elif s>_HIGHSTEP: s = _HIGHSTEP
		if self.zstep is not self.step:
			step, power = ControlFrame._stepPower(self.zstep.get())
			**#zs = step/10.0
			zs = s**
			if zs<_LOWSTEP: zs = _LOWSTEP
			elif zs>_HIGHZSTEP: zs = _HIGHZSTEP
		else:
			zs=None
		self.setStep(s, zs)

So ZStep never exceeds HIGHZSTEP but below ZStep is always the same as STEP.

Harvie, 2018-09-18:
Maybe we can add something like "SAME" option into zstep drop menu, which would mean that zstep will remain same as xystep no matter what.

Harvie, 2018-09-18:
[image]

Guenni75, 2018-09-19:
Ok I try it. Thanks

Guenni75, 2018-09-19:
Ok, the "SAME" function works.
But now he is ignoring the HIGHZSTEP!
And it feels a little bit strange that i can not see the current ZSTEP Value.

einencool, 2018-11-02:
Hello,
I‘ve got a question for the „same“ function.
On my external keyboard I‘ve the following function on the keys 1-3:
1: step = 0,1mm
2: step = 1mm
3: step = 10mm

When I press the buttons 1 or 2 the steps for XYZ are equal, but when I press button 3 my XY steps are 10mm and my Z step will remain / be set to 1mm 

How can I change this, the for all three buttons the steps will be set as usual?
It‘s nice to have the possibility to set XY and Z steps to different values, but I would like to have 10mm for Z as well.
Greets Chris

Harvie, 2018-11-05:
I am not really sure. Maybe it's not possible to set Z step so high? IIRC there's some step limit.

einencool, 2018-11-05:
With the "+" key I can change my Z-steps to 100mm (it's the max travel)
For now, I change the steps for all axis to 1.0mm and press 10x "+" key, so I can travel with 10mm on every axis...

In my old Versoin of bCNC where I only have one value for all axis there was no Problem with this.

Maybe you know where I can search for it.

Harvie, 2018-11-06:
There's "XY" setting in the left dropdown. Please try it.

einencool, 2018-11-06:
Hi @Harvie,

I've tested it last time too, but now I make some Screenshots :-)

When I choose the "XY" setting and THEN press any key 1 / 2 / 3 on my external keypad, the steps change to 
0,1 / 0,1
1,0 / 1,0
1,0 / 10

Before:
[image]

After:
[image]

Harvie, 2018-11-07:
What if you pres * and / keys ?

einencool, 2018-11-07:
Hi @harvie,

this is sadly not an option for me, because I mapped the keys to other functions like Goto X0Y0 and Homing etc. 

Because I don't need the buttons for multiple ir divide the step width.
But I tested it for you, and then the following steps happen:
Z   /   XY
0,1 / 0,1
1,0 / 1,0
10 / 10
10 / 100
1,0 / 10
0,1 / 1,0
0,01 / 0,1

this happens when I press * and / several times...

Where can I find the commands for the 1 / 2 / 3 keys on the keypad?
There should be the answer, or am I wrong?

Greets
Chris

Harvie, 2018-11-08:
What is last version of bCNC which worked as you expect?

einencool, 2018-11-08:
Hi @harvie

It was I think in August, but there was only the possibility to set all steps to one value.

I will make a screen shot in the evening, then I can show you the behavior with the old version.

That was the last version before you added the hardware start button. Then I updated  to the last two versions.

Greets Chris