vlachoudis/bCNC #938: Stopping Autolevel Causes a Freeze

CoolNamesAllTaken, 2018-08-16:
Whenever I stop an autolevel operation, when I run a new one the machine moves to the first probe point and rapid-moves to the surface, stopping on top of it (z=0).  bCNC becomes unresponsive to unlocks, resets, movements, and further probe commands after this.  A full program reboot is required to fix the issue. Has anyone else experienced this?

CoolNamesAllTaken, 2018-08-16:
Upon closer investigation it looks like I'm getting a GRBL Alarm 4.  My GRBL settings are as follows.

Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
Grbl 1.1f ['$' for help]
$0=10
$1=255
$2=0
$3=2
$4=0
$5=0
$6=0
$10=19
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=25.000
$25=1000.000
$26=44
$27=4.000
$30=12000
$31=0
$32=0
$100=40.000
$101=40.000
$102=80.000
$110=8000.000
$111=8000.000
$112=3000.000
$120=250.000
$121=250.000
$122=250.000
$130=264.000
$131=264.000
$132=69.000

My standard procedure is to probe a point, set it as z=0, then run the autolevel scan.

I've had this problem crop up on the first point, and the last point of the auto-level sequence.

CoolNamesAllTaken, 2018-08-16:
It looks like this error may have been caused by starting the probe height too low (0.25mm), causing the head to trigger before it started lowering.  I raised the probe height to 0.5 and soldered a 100nF capacitor Z-probe, and things seem to be working ok now.  Still quite bizarre that it created such an aggressive crash mode (I was forced to reboot the CNC machine and reset the serial connection to get things to work again once they crashed).

Harvie, 2018-10-30:
Duplicate of #642

closed 2018-10-30 by Harvie